KimJorgensen/KungFuFlash #2: Pin mapping

kke, 2020-01-25:
I've been trying to reimplement this but it's next to impossible to route due to the weird pinout of the STM (which in it's datasheet states it requires a 4 layer board). 

Making any changes to the current design will make it very difficult if not impossible to route.

I was just thinking, would it make sense to change the pin mapping in this relatively early stage so that it would be easier? For example by having all the D0..7 on one side of the board in the same order as they are on the expansion port, and something similar for the A0..15.

KimJorgensen, 2020-01-25:
The pin mapping was deliberately made this way to make the bus handling as fast as possible. The timing is close to the limit and additional reads or bit shifting due to a different pin mapping would limit what could be emulated by Kung Fu Flash.

I did read that it requires a 4 layer board but chose 2 layer anyway to keep the cost down. That was quite challenging but not impossible as you can see.

So what are you trying to achieve since you want to re-implement it?

kke, 2020-01-25:
To use the parts I have a zillion of from other projects where the minimum order amount is 50-100 and I needed like 4, and also because laying out PCB's is kind of fun.

The bit shifting makes a lot of sense. Perhaps this was a silly thought :)

closed 2020-01-25 by kke