KimJorgensen/KungFuFlash #4: SX-64 not working.

MichaelTroelsen, 2020-01-26:
Kung fu not working on sx-64

KimJorgensen, 2020-01-27:
I guess you are the first to test it on a SX-64, thank you. Is it the PAL version?
What does not working mean? Do you just get a black screen at power-up, or is it unstable?

kke, 2020-01-28:
https://comp.sys.cbm.narkive.com/aBgNh4QT/why-all-c64-cartridged-dont-work-at-sx64

> The SX64 cart port is a bit on the edge. It's connected
to the main board with a flat cable which is too long for its own good.

> There are two problems: not enough power, and too much interference on
the port extension cable. Some carts just use more power than the
SX-64's PSU can supply, notably REUs and the SuperCPU. The ribbon
cable used to extend the cartridge port to the top of the SX-64 case
needs to be replaced on some machines. Find a high quality shielded
cable and you shouldn't have any more problems.

smdprutser, 2020-10-25:
I did some quick testing on a European SX64 and it does work.

However I did replace its ribbon cable in the past (I made a replacement EXP board) with a more recent ribbon cable with 28AWG wire in it. Dunno how much AWG the original cable is, but it felt more flimpsy then the new cable.

KimJorgensen, 2020-10-26:
@smdprutser Great. Thank you for testing

zymurgic, 2021-07-07:
Confirmed as working on a PAL SX64 with PLA20V8 replacement PLA fitted. Aside from the PLA, it's a stock PAL SX64.

Gun-neR, 2024-01-10:
I'll add my two bits even though it is closed, as reading here I suspect that my issue may also simply be the hardware and not the cartridge???...

- SX-64 NTSC - MFG June 1984 - In really great condition, this KFF is the first cartridge that I have noticed any issue with, but then I mostly use an SD2IEC with Fastboot.
- Kung Fu Flash (from TFW8b) currently running v1.49 (I have upgraded FW twice since initial purchase late last year, hoping that might help).

My KFF generally works great when it does... If I manage to get something running, then all is good :D

However, more often then not :(  when I try to switch to another program (no particular program causes the issue), the KFF has a "fit" with both Menu and Reset LEDs going into a fast strobing mode, the drive LED lights up and drive spins, and the screen "scrambles" (almost like video memory mapping issue) or goes some blank colour (white, yellow, etc.).  Once that happens, the issue is persistent through resets and power cycles... 

Only solution so far has been to shut off and wait awhile (no idea how this helps as it is not an apparent heat issue... often crashes right away on first attempt to use after powerup or hours later when changing to another program.... or it just works... roll of dice.  

Sometimes pulling the SD card and erasing the kungfuflash.dat file is required to reset it... or perhaps just a placebo :P 

It has been frustrating enough that I often regret purchasing such a costly item and just stop using the KFF for extended duration.

Meanwhile, I guess I will look into the ribbon cable, and perhaps adding a physical drive selector option as I wonder if the built in drive "sharing" device 8 with the KFF can also be an issue??  (Changing the KFF to 9, assuming that doesn't cause issues with programs, doesn't seem to stay saved that way, so I can't tell if matters).

KimJorgensen, 2024-01-10:
Yes, in the future please open a new issue and refer to the closed one instead.

The Kung Fu Flash cartridge enters this "fast strobing mode" until a valid C64 clock signal is detected. In this mode the C64 will be held reset which could explain the drive behaviour and the scrambled screen.

Deleting the KungFuFlash.dat file or changing the drive device number does not affect the detection of the C64 clock signal in any way. However, a bad ribbon cable could indeed be the problem.

I do understand your frustration, but please keep in mind that this my personal project that I have shared completely free of charge. Me and others who answer questions here do so in our spare time without receiving any money from the sellers of these cartridges.

Gun-neR, 2024-01-10:
@KimJorgensen  Sorry for the poor etiquette, I had actually considered opening a new one, but hesitated as I thought the "solution" was likely the cable anyway, thus thought "why duplicate the same concept" :P

But thanks for the detailed info.  That makes sense, so I will pursue a cable upgrade option and offer feedback here whenever done.

As for my frustration, my bad on the lack of clarity... I had messaged the seller (who recommended it to me for the C64 after I had success with his Penultimate carts for my VIC-20) many times with my issues and wondering of what might be causing them... but never so much as a peep back :( ... So all frustration was actually directed in that direction, not here ;)  

Thank You for this creation!!  As I said, it does work well when it does, and is a great addition for overall usability of a 40 year old relic.  And has actually been instrumental in my putting more efforts and $$$ toward some recent upgrades and repairs (colour matched 3D printed endcaps and a new KB membrane), so I think now a cable, heatsinks and possibly a device selector option for the drive is in order.

KimJorgensen, 2024-01-10:
@Gun-neR Thanks. I hope that a new cable will fix the issue, but feedback is welcome either way.