KimJorgensen/KungFuFlash #5: Expert Cartridge

MichaelTroelsen, 2020-01-26:
Expert Cartridge not supported

KimJorgensen, 2020-01-27:
Correct. It is not on the Supported Cartridges list so I'll label this as an enhancement

modeler, 2020-11-30:
I would love to see the Expert added, but it wasn't like most other freezer cartridges. It had a three-way toggle switch (on / off / program) and the freeze function was activated by the RESTORE key. It would have to be controlled through the interface and that can't be a straightforward task.

KimJorgensen, 2020-12-01:
I also got a version with a ESM button. So yes, this and the three-way toggle switch must somehow be emulated with the buttons on the Kung Fu Flash

modeler, 2020-12-01:
It might not be necessary to emulate the programming mode, since it is possible (and nowadays, desirable) to create .CRT images of the Expert with its various 8 KB programs already loaded. Perhaps the Kung Fu Flash's 'special' button could be used to toggle the 'on' and 'off' function?

Expert emulation would be a nice feature. It's not implemented in Ultimate64 and it is still incomplete in VICE (the cruncher crashes the emulated machine and writes garbage).

mwedmark, 2020-12-16:
Sorry to clutter the thread. I actually ordered a ExpertCartridge back in the days and already had a good 8K bank ready to be downloaded into that programmable spot. Sadly, it never was delivered..

smdprutser, 2021-01-13:
I have made a pull request with Expert support. see more info here: #75

modeler, 2021-01-13:
Great work! I have built it and tried it, I am able to load an Expert Cartridge image and drop in and out of the emulated cartridge using the RESTORE key, use the machine monitor and such just like the real thing. Found a couple of issues though:

1. I was unable to program the cartridge with a new payload, although I did use the recommended settings and exited the menu with F8 it crashed when I hit reset after programming.

[image]

2. The 'crunch' function didn't work for me, I was able to load a program and freeze it, but the resulting program crashed while decompressing. Looks very similar to what happens in VICE currently (old bug but still present in v3.5): https://sourceforge.net/p/vice-emu/bugs/939/

Thanks for your efforts thus far, it's impressive to see Expert support in any form!

smdprutser, 2021-01-14:
I had sometimes that programming didn't work (mostly after an reset), but usually after pressing the menu and retrying it works and keeps working. Always though it was because I'm devving on an SX64 which give some trouble with carts.

That crunch function does work on a real C64? I have an Expert on it way but cannot test it yet. Will look more into it when it arrives.

modeler, 2021-01-14:
The crunch function doesn't work for me using a real C64 (tried a couple of different ones, got loads here) and Kung Fu Flash emulating Expert. Same result as in VICE, just crashes at the end of decrunching every time.

I have a real Expert Cartridge here, which works fine. Same C64, same Expert software (V3.2R) and same game.

smdprutser, 2021-01-15:
I'll dive more into the code this weekend. Have some ideas why it is failing.

BTW Does the loading of the payload work for you now? (by pressing the menu button and then F8 a couiple of times repeated) Will look also in this problem too.

modeler, 2021-01-16:
I had another go at programming the cartridge and was able to get it to work fairly reliably by doing the following:

Menu -> F8
Load up Expert module
Reset
Menu -> F8
Reset
(Trilogic splash screen appears)

I couldn't quite get my head around how it works, I must have been hitting RESTORE a bit too soon the first time I tried it. I'm impressed that it does work though, even with the V4 'Multi-Menu' release of the software. The crunch function (renamed to 'backup' in V4) still doesn't work for me though.

Please let me know if you need any further info / testing, happy to help if I can!

smdprutser, 2021-01-24:
I got the booting sorted now. There are sometime stray read or writes to the IO1 area which triggers the cart to switch to idle mode instead of programmode.

Regarding the crashing after saving a stopped program I suspect some timing issues as some RC circuitry in the cart and ATM I switch instantly. If someone could lent me an opened Expert cart or can provide the timings I can implement them.

modeler, 2021-01-24:
Not sure I can offer any help but if I can, I will. The innards of my Expert look like this:

[image]

They scratched off the IC markings. Someone reverse engineered it and shared a schematic here, if it's any use?:

https://people.kth.se/~e93_msi/c64/expert.html

smdprutser, 2021-01-26:
Thanks for the Photo. I know this page and my code code heavily depends on this :)

However if I use the values of the resistors and caps I get ridiculous delay times. From your picture I think all resistors are the same. I think they are orange-orange-red-gold? 

I'm diving in tomorrow and try to split my pull request into two as Kim wants

modeler, 2021-01-26:
Yes, all four are orange-orange-red-gold (3.3 kΩ).