KimJorgensen/KungFuFlash #6: NTSC not working

kcasteels, 2020-01-28:
I know that NTSC isn't currently supported. Will it be supported in the future?

KimJorgensen, 2020-01-28:
I certainly hope so. The main reason for it not being supported is that I don't own a NTSC C64.

The firmware already measures the CPU frequency and refuses to start if it is inserted into a NTSC machine (I hope). So the idea was to use this to auto detect which machine it running on and adjust the delays accordingly.

It could be that firmware optimizations also are required for the most time critical parts (e.g. C128 2MHz mode) as the NTSC version runs at a higher CPU frequency.

If somebody would like a go at this I'll be happy to accept a pull request

rihardgDev, 2020-09-26:
Hello, first of all I would like to express admiration for this wonderful project. I've been using Kung Fu Flash for couple of months and I must say I use ti much more than my Pi1451.  I've wanted to help spread the world about this wonderful and versatile cart so I've created a blog with hopefully easy to understand instructions on how to build it: https://8bithardware.wixsite.com/website/post/kung-fu-flash (you can link it, if you think it's useful). I've been promoting it via Facebook groups and the general opinion is very good. With one exception, the NTSC part of the community feels left out. I think that adding NTSC would benefit the project greatly. With bigger audience, you automatically attract more developers and new features bring in even more new users and so on... 

I know that the main reason for lacking the NTSC is that you don't have a NTSC C64, so i think it might be worth to try to find a developer who has one. I'm pretty sure that there must be some talented people from NTSC  world that would be interested in implementing this feature. To ease this process, I would propose that a "TODO" documentation is made, describing what in general needs to be done and how it's implemented on PAL version. I think we all agree it's easier to start if there is at least a basic documentation/idea whats to be done. 

One gentlemen on C64 Facebook group even said it can't be done because it's already running on the edge of the CPU capabilities etc... 
With a short document describing the basic requirements and where in the current code is PAL implementation this kind of dilemmas would be resolved and I'm pretty sure someone would start implementing it.

I really think this is successor of the Easy Flash 3 and a drastic enhancement of its capabilities and it deserves to be widespread! Thank you for your gift to the community!

tfw8b, 2020-12-06:
> I certainly hope so. The main reason for it not being supported is that I don't own a NTSC C64.
>
> The firmware already measures the CPU frequency and refuses to start if it is inserted into a NTSC [...]

Do let me know over at tfw8b if we can do anything to help. I've got a few NTSC machines knocking about.. and I might even have an NTSC 128 here somewhere ;D

@futurewas8bit

KimJorgensen, 2020-12-17:
A new firmware with NTSC support has been [released](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.17). This version has only been limited tested so I would appreciate help to test this release and some feedback.

Icelvlan88, 2020-12-19:
All I get is a black screen. Does it work without an sd card and just firmware?

KimJorgensen, 2020-12-19:
It will not work without an sd card but you should still be able to see something on the screen. Is the LED lit?

Icelvlan88, 2020-12-20:
Unfortunately I trace lifted for the LED so I took it out and cut it...

Icelvlan88, 2020-12-28:
Which files need to be on the sd card?

KimJorgensen, 2020-12-28:
Only the files you want to run need to be on the SD card, nothing else.

closed 2021-04-11 by KimJorgensen