KimJorgensen/KungFuFlash #10: SID File Support

djtulan, 2020-04-06:
I tried out my Kung Fu Flash yesterday and filled my SD Card with a lot of software used on my Ultimate 1541 II. By chance I came across a folder with SID files and I was wondering, if SID file playback support would be possible on the Kung Fu Flash too?
If there is space left, it would be a nice enhancement.

KimJorgensen, 2020-04-06:
Yes, I guess that would be possible and a nice addition.

djtulan, 2020-04-06:
You could convert the SID file with this code, an load it as a RPG file.
https://csdb.dk/forums/index.php?roomid=14&topicid=103013

mwedmark, 2020-12-07:
I'm looking at this right now. I have 2 possible ways, maybe try 1 first which I already have going as a standalone VICE/Breadbox

1) Use C64 to fix file, minimal support which will only work for max 18k files and only files that lies within tight memory bounds, use my own strange code which will work for some. Minimal work at STM32 side but limiting.
2) Use PSID64 code which is one of the better methods for doing SID=>PRG, compiled for STM32 and let the ARM build the file to PRG in memory and sent it to C64 and run it there. When compiled for PC/WIn the executable becomes 182KB and I guess that is a pointer to the size that this would need to be in ARM too.. How big can such a STM32 plugin become? Could it be added as a seperate plugin module in a special directory of the SD-card? Any ideas?

KimJorgensen, 2020-12-08:
It is my assumption that an ARM build of PSID64 will be a lot smaller than the Windows executable but there is not much room left in the flash (around 4kB). Another issue is the license.
Currently the firmware has no plugin support and all code is statically linked. I guess that needs to change if we want to keep adding features to the Kung Fu Flash.

mwedmark, 2020-12-08:
OK that kind of what I guessed. For the fun of it, I'll start by adding a minimum SID info view so I learn. You decide to bring it in or not.

mwedmark, 2020-12-10:
First attempt at handling SID file and showing info! Feedback anyone?

[image]

[image]

KimJorgensen, 2020-12-10:
I think it is fine as learning experience but wouldn't personally prioritise adding file info screens to the launcher. With SID files I would just display the song info in the player.

mwedmark, 2020-12-20:
Status update:
I've still trying to get a simple custom KFF SID player working and after some backlashes I'm currently stuck (some strange handling of char arrays for STM32). I think I've got an alignment issue but I'm not sure.

If I include the *.SID file (PSID) as a binary file in the assembler, it works. But if I try to load my SID play code PRG (around $1000) and than on top of that (higher in mem) using STM32 load from SD-card it just never works. What kind of tools do you guys use for debugging? I have a STLink but not sure how to use it. I'm yet to start using debugger because I'm currently not using an IDE, just building from command line. Would be great to be able to use breakpoints, watches and everything I'm use to in Visual Studio.

KimJorgensen, 2020-12-21:
It is possible to setup a ST-Link GDB Server and use that for debugging. However, you can't really stop execution on the STM32 (use breakpoints) as this will probably crash the C64 if it tries to access the cartridge.
I'm using the USB port for "printf debugging", see print.h on how to enable logging. You may also need to comment out `usb_disable` in main.c depending on where you want to log.

medzes, 2021-02-01:
I think it would be great if SID player functionality could be combined with BackBit's BTT file support. This has SID tunes and screenshots in Koala format which could be displayed as a preview.
But the code space available in the static 48kB is too limited at this time to allow such large code. I think it would make sense to have a plugin system allowing the menu ARM code to be loaded from SDcard, as an alternative to the built-in menu as Kim suggests.

Fabuseta, 2021-09-12:
Any progression towards SID support?

mwedmark, 2021-09-12:
I'm sorry to say, no progression for me. I have the whole toolchain working, did a inital "View" function that read SID-info and showed them on-screen. I've also written my own simple (naive) SID-player function which works pretty good. It's missing relocation and probably other functions. The problem that stopped me, was some kind of tool-chain bug/behaviour which made a hard-coded SID-song work but a dynamic one loaded from SD-card not work.. without knowledge about ARM debugging tools I've currently lost interest. If someone wants to have a look I've could create a fork.. I may come back when again

Sgw32, 2023-02-14:
> First attempt at handling SID file and showing info! Feedback anyone?
> 
> [image]
> 
> [image]

Can you share code? I plan SID emulation using KFF. Thanks!

mwedmark, 2023-02-15:
I guess we've been talking on the other projects about PicoSID/UltiSID! :-) 
Yes, I have no problem sharing the code but it is in a crude state. It wont help you at all managing a SID emulation with KFF sorry to say, as it does not include STM32 doing any such thing here. The only thing I did was to do add stuff to the menu, a new SID file type(without working playing) and possible to show info. It also supports entering "USB only mode" after 10s waiting for SD-card. I though that was a little useful when you use it for developing sending stuff over USB. I will create a fork and just ping if there is strange stuff. There are some other tests being made in the code as well.  Sorry to say a bad merge situation was also present when creating the fork and currently I have not an easy way to build and test it. Would be really cool if you could find the problem to why my SID-play routing didn't load correctly when trying to transfer SID-file info from ds-card via STM32 to C64 memory. Tried some PRAGMA pack but still had some kind of align issue maybe between byte/word/dword..

Sgw32, 2023-02-15:
> I guess we've been talking on the other projects about PicoSID/UltiSID! :-) Yes, I have no problem sharing the code but it is in a crude state. It wont help you at all managing a SID emulation with [...]

Thanks! I'll try to test it.