KimJorgensen/KungFuFlash #11: BOM mistake

djtulan, 2020-04-23:
Hi!
In BOM there is a little mistake.

_Tactile Push Button Micro Switch - Use 6x6x6mm with TFW8b button caps or 6x6x9mm with 3.1mm A101 Plastic Switch Button Caps_

If somebody want to use the A101 Plastic Caps a 6x6mm with ~11mm height ist needed.

KimJorgensen, 2020-04-23:
Thank you for reporting this. That should be fixed.

djtulan, 2020-04-23:
The button height should be exactly 11mm. From the PCB to the top. I measured it again.

KimJorgensen, 2020-04-23:
Thanks. I'll update the BOM with this.

djtulan, 2020-04-23:
[image]

KimJorgensen, 2020-04-26:
The BOM file has now been updated with the correct size of the push buttons.
Thank you.

closed 2020-04-26 by KimJorgensen