KimJorgensen/KungFuFlash #12: Stability problems with replacement PLA's

patzik, 2020-04-23:
With a (modern) replacement PLA, sometimes when mounting a d64 and loading a prg from it, either from menu of after a reset, the C64 comes with an error.
The errors I have seen are "out of memory", "drive not found"
Tested PLA's are:
DodgyPLA https://github.com/desaster/c64-dodgypla
MattisLind https://github.com/MattisLind/82S100replacement
PLAnkton

KimJorgensen, 2020-05-23:
According to @dikdom KFF works fine with the PLA20V8 PLA replacement.

KimJorgensen, 2020-06-17:
Could you please retest with version 1.09?
That version has been tested on my C64 ASSY #250407 with DodgyPLA, PLAnkton and J-PLA.

patzik, 2020-06-21:
Tested with 1.09, works like a charm. Great work!!

KimJorgensen, 2020-06-21:
Great, thanks for testing

closed 2020-06-21 by KimJorgensen