KimJorgensen/KungFuFlash #13: export .D64 to floppy enhancement

ConradsenDK, 2020-04-25:
would be cool if it could copy a image to floppydisk :-)

KimJorgensen, 2020-04-25:
You can do that via USB and ef3utils. But yes, I would be cool if you could do that directly from the SD card.

dikdom, 2020-04-28:
... with the possibility of fast data transfer (i.e not taking 15mins to copy an image file to disk).. as a second iteration maybe.

Tissemon, 2025-05-06:
The later versions of DraCopy seems to support copy to real floppy. I saw that code from DraCopy was already being used in KFF so maybe these new features could be added quite easily? https://github.com/doj/dracopy

neunewcomer, 2025-06-06:
Or there’s a program called KungFuFlashCopy made by rh70 (you can find it on Forum64). Transfers from SD to floppy, and vice versa, at a more decent speed than DraCopy.