KimJorgensen/KungFuFlash #14: Kernal support

dikdom, 2020-04-27:
Hello,
I was checked this cart (fantastic work, btw) and compared with EF3 and I realized that this one doesn't support the KERNAL switch, meaning that you could select a KERNAL image to replace the original one.
Is there any particular reason for that or just simply missed from the initial feature list?
Sorry for my ignorance I don't know how big deal is to implement that, so it may be a noob question..

KimJorgensen, 2020-04-27:
After being asked about this on forum64 I have looked a bit into it, but I think it is good to keep track of the issue here.
I have tried to implement the method described in "[Designing a Real KERNAL Cartridge](https://github.com/KimJorgensen/easyflash/releases/download/EF3-FC3/kernal-cartridge.pdf)" by @skoe but has not been successful so far. Pulling A14 low from KFF does not seem to work, maybe the timing is off or maybe the signal isn’t pulled low enough.
More investigation is needed.

dikdom, 2020-04-27:
Thanks for your reply and thanks for sharing a the doc, I haven't seen it earlier!

In the KFF there is only a 8MHz clock (oh, OK, you may have PLL), on the other hand I don't understand why it doesn't cause any conflict (HW failure in time) to pull down A14 externally while the CPU may drive it to high.. 
And yes, the doc doesn't mention how does the EF3 pull down the A14 safely (below 0.8V I guess) without burning the CPU.

Thanks for your time!

KimJorgensen, 2020-04-27:
The ARM processor runs at 168MHz but there are delays involved reading the bus, checking for the correct condition, writing back to the bus etc. It will almost always be slower than when implemented in hardware or a CPLD as EF3.
If you check the schematic for EF3 there is no hardware protection on A14. The signal goes straight to the CPLD which can sink up to 70mA if I read the datasheet correctly.
I think the key to not damage the CPU is that A14 is only pulled down a fraction of the clock cycle and this is only done after a write to address $0000 or $0001.

skoe, 2020-04-27:
@KimJorgensen This is correct. These are the points that allow this to be done:

- NMOS pull-ups are quite weak. They can't source much. However, the closer the output comes to 0V, the stronger they get.
- It's a fraction of a clock cycle only, as you said.
- It happens at most every few clock cycles, usually much much more seldom. So the FETs can't heat up.
- Actually, similar effects happen in every machine: When the chip select changes from one IC to another, not all of them have the same speed, so the data bus may be driven from one IC already before the previous IC lets it go.

About the timing: It must be really very accurate I remember. It was many years ago, details I can't remember unfortunately.

Hope this clarifies your question. Note that I didn't have a look at this project but got a notification from github :)

dikdom, 2020-04-28:
Hello,
checking the STM32f405RG spec I read a note (DocID022152 Rev 8, page 59):
"PC13, PC14, PC15 and PI8 are supplied through the power switch. Since the switch only sinks a limited amount of current
(3 mA), the use of GPIOs PC13 to PC15 and PI8 in output mode is limited:
- The speed should not exceed 2 MHz with a maximum load of 30 pF.
- These I/Os must not be used as a current source (e.g. to drive an LED)."

1) Can it be the cause that you cannot change at proper speed the /GAME and /EXROM lines (max 2MHz seems not enough if you need to pull them down for 80ns or 160ns)
2) PC13 is a current sink, so driving a LED is not a problem here, I guess.
3) I have one more question thought it seems trivial (but not for me, sorry). How do you know that the ARM chip is 5V tolerant?

KimJorgensen, 2020-04-28:
@skoe Thank you for taking the time for the clarifications and for your excellent document. If you can find the time, please consider putting the newest version on your website so others can find it too. From what you are saying I assume that the timing is off but I really need to verify that with a logic analyzer.

@dikdom
1. That could be. I haven't really spend any time investigating this
2. Yes, using it as a sink works fine. I'm not sure that is documented anywhere
3. In table 7 in the document that you referenced there is a "FT" next to each pin that is 5V tolerant

It was actually quite a puzzle to map all signals to pins, as they all are in use. Not all are 5V tolerant, some have a different use when running the bootloader which could damage the C64, and not least, it should easy for the software to read the address and data bus without having to do bit shifting.

dikdom, 2020-04-28:
3. I just should have read 2 more lines... :) The note I sent earlier is in the same legend of the table 7 you are referring to.. 
I can deeply understand the puzzling part. I'm not an expert, designing a PCB for me is let the autorouter do the job (under a few MHz there should not be too much harm)..

So in this case I'll probably wait sometime with the ordering the PCB and I hope you will come up with a good solution... :)
I would multiplex the /EXROM and /GAME with the SD reader pins (you don't use the SD and the C64 at the same time, I assume) and disconnect the C64 lines with a 4066 like switch when using the SD. 
But the SD lines are the other side of the MCU and of course I don't know if the multiplexing could be done at all, and the bit-banging... and.. a lot more can be considered I don't even think about.

medzes, 2021-02-01:
HI all, 

I've successfully been able to get kernal mode working exactly as skoe documented in his excellent article in KungFuFlash, using the DEVEBOX prototype I have. The tricks needed to do that are primarily a lot of work getting the timing right and checking the oscilloscope.
Now I don't own a regular KungFuFlash cartridge so I cannot test / check the timing for the regular cartridge and I can also not check the slope speeds of the signals for the regular PCB rev2 version. I have PA14, EXROM, GAME on a regular GPIO. (not on PC13,14,15). But I expect the same tricks below would work on PCB rev2 as well.

What was needed to get it to work:
- Normal 5V tolerant use for EXROM, GAME means open-drain config. However, this means both EXROM and GAME have a rising edge slope that is far too slow to reach the timing mentioned in skoe's document during the A14 tests.
- It is also likely in some corner cases like some C64 PCBs or some specific address transitions causing voltage drops, the half-cycle delay between releasing GAME and the begin of next CPU cycle may not allow GAME to be pulled up fast enough by the resistor inside the C64 alone.
-  So, a trick is needed to momentarily set EXROM/GAME to normal high output with a 4 cycle wait (in my tests) to let EXROM/GAME reach 3.3V, before releasing them to let the pullup resistor in the C64 handle the rest.
- We cannot leave EXROM/GAME in output mode all the time as that would mean the C64 pullup resistor for EXROM/GAME will increase the 3V3 on the ARM IC. This may lead (if the arm is not consuming enough power) to a 3V3 above 3.6V, which damages the ARM.
(I have however programmed EXROM/GAME to output fixed by mistake though, and that didn't blow up anything - then again the 2 LEDs on the DEVEBOX alone draw a few mA alone, probably sufficient to keep the 3V3 below 3.6V)
- A14 is pulled low very fast after it is set to output, and rises back high very quick too after release. No special stuff needed.
- The A14 check loop requires more time before the interrupt occurs than the existing cartridge timing allows. I've re-implemented a specific handler from scratch with earlier interrupt timing to fit the tighter timing. 
- The A14, EXROM, GAME OTYPER/MODER toggling are highly dependant on exact cycle delay inbetween the actions. Using NOPs didnt allow sufficient time for all functions so I had to move some logic around As a result, the code is C but it is so sensitive to the compiler / optimizer and exact constants, it might as well be assembly.
- The 48kB code size is very full, so I switched to -Os and some always_inlines. This allows the kernal code to fit.

I will upload my changes to my fork after I have been able to run some tests that perform $00 / $01 tests. Does anyone have some suggestions what C64 software to use (that changes $00/$01 a lot and is very sensitive) to test the functionality properly?

skoe, 2021-02-01:
Congratulations :) Regarding the tests: I tested the mechanism with hand written assembly loops, because this made it easy to check everything on a logic analyzer. And when this worked, I run all kinds of programs :) 

Did you also implement the mechanism from chapter 2.2 Optimizations to avoid pulling down A14 in every cycle? The will reduce the stress on the bus drivers significantly.

medzes, 2021-02-01:
Hi skoe,

Thanks for that insight. I guess I could also write and use some handcrafted loops, will take some time to integrate that.

Yes, I have also implemented the mechanism to avoid pulling down A14 every cycle as you suggested. In fact there are 3 separate interrupt service routines, currently using a state variable. It could be optimized using 3 separate ISR vectors, that would make the A14-check ISR with its tough timing more compatible with background usb or other activity by switching to the less restrictive timing (without the A14 check) on the fly.

I notice in the idle kernal loop, there are occasional $01 writes happening at some kHz rate. So from this I believe its all working correct. The interrupt timing of the arm is fortunately very exact so far. I may need to support a 1 or 2 cycle ISR delay inaccuracy - I use a C64 VIC cycle alignment style loop to get to 1 cycle accurate timing based on the Timer capture value. That part was very fun - almost exactly like using the 6526 timer to get in lockstep with the VIC, but then on the ARM with phi2.

medzes, 2021-02-11:
Hi, I've been able to find the time to clean up my code,  test it a bit and make it ready for others to check out. You can find the kernal mode on my fork. It can be compiled for PCB rev 2 or DEVEBOX using a define in the Makefile.
This version is not yet rock steady, but it does work for many code and games. 
The code supports two mode:
- Select a kernal ROM file. Drops you into a normal basic screen.
- Place a file called "kernal.bin" on your SDcard. If this file is detected, it will be used to start PRG and disk based programs from the easyflash menu.
Please note I have not checked the PCB rev 2 version as I don't own one. 
If you are interested to test, please be aware the constants of peripheral base addresses are different which can make the compiler create code with slightly or even big timing differences. It is best to compare the generated kernal_handler code in the build/main.lst by hand in case of doubts.
Also it is a good idea to check with an oscilloscope the timing of the A14 check and the $de02 write check. The kernal.c file contains some notes how you can do that and what timing to expect.