KimJorgensen/KungFuFlash #15: BOM question

dikdom, 2020-05-04:
Hello,
could you please put a photograph about the cart from the USB socket side as I'm having a hard time figuring out what kind of USB socket is needed (tbh I'm afraid of ordering from Ali because of the arrival time of the free shipped orders...). So. On the kicad PCB file it seems that the bottom pin is the pin 1, +5V from USB, top pin is GND, pin 5. So, checking the pinout of the microUSB socket from the wiki (https://en.wikipedia.org/wiki/USB_hardware), the socket has to be laid on its longer side (shorter side of the microUSB trapeze is UP). But checking the picture on the ali link you specified in the BOM, it is quite the opposite. So which one? 
I assume you already used the USB interface of KFF, so it works that way you have in your cart.
A picture would help.

KimJorgensen, 2020-05-04:
Hi,
The bottom pin is pin 1 and the socket is laid on its short side, just like shown on the wiki (https://en.wikipedia.org/wiki/File:USB_Micro-B_receptacle.svg).

Here is a picture:
[image]
And yes, the USB interface works fine.

Hope this helps and please let me know if you find a suitable replacement.

dikdom, 2020-05-04:
Thanks, then I mixed the front and back view... This picture is not so straightforward.. https://en.wikipedia.org/wiki/USB_hardware#/media/File:USB_2.0_connectors.svg
I'm glad I asked! :)

For the micro USB B socket I hope this will work for me: 10118194-0001LF from Amphenol.
For the micro SD card socket I'm targeting this: 112J-TDAR-R01 from Attend.

Some months and I will let you know the results! :)

closed 2020-05-04 by dikdom