KimJorgensen/KungFuFlash #16: troubleshooting

dikdom, 2020-05-19:
Hello,
a little help would be appreciated... 

So, PCBs arrived, most of the parts are pre-soldered. USB and SD slots, switches and jumper headers are soldered. Checked the USB slot with a cut USB cable, each wire is OK (+5V, GND, D- and D+, no shorts)

- firmware (v1.08) is uploaded ('Downloaded' in the terminology of dfu-utils) and read back ('Uploaded') and verified. It is OK. 

Q1: is it true that the computer should be able to see the board if I press all the three buttons at the same time? I have to press them at once in order win10 be able to see it.

Q2: no LED activity, when the board is connected to USB. Is it normal?

- OK, fw programmed, when I try it with C64, the LED is lit but the screen remains black. With an SD card, without an SD card.

I tested the SD slot's pins and they aren't shorted. 

Q3: Any idea where to start the debugging?

KimJorgensen, 2020-05-21:
Hi,

Q1: There is no functionality that requires you to press all buttons at once. It may be necessary to press the reset button after shorting BOOT0 and +3.3V to make the microcontroller enter the built-in bootloader. When the bootloader is running the device should be detected by windows.

Q2: Yes, there is no LED activity when the bootloader is running.

Q3: If replacing the resistor networks doesn't help then try it with a different C64, preferably one with an original kernal and PLA.

closed 2020-05-21 by dikdom

rickard-von-essen, 2020-08-11:
@dikdom was this resolved by replacing RN1/RN2?

dikdom, 2020-09-07:
@rickard-von-essen yes, it did the trick.