KimJorgensen/KungFuFlash #17: Questions

dikdom, 2020-05-22:
Hello,
finally I managed to fix my cart and it starts, it works. Most of the things are abvious and fine but I have some questions:
1) when I turn on the cart or reset sometimes it starts with the normal C64 boot screen and sometimes it goes directly to the KFF menu. What does it depends on?
2) I have found the F5 configuration option. What does that exactly mean? Is there any longer description about it somewhere? 
3) Final Cart III and FC III+, the freeze menu is strange, though it works. There is some flash on the upper menu and on the drop down menu, but it is responsive to the keys and works. I will attach a screenshot later (motherboard 250407, normal PLA chip, normal VIC-II). With a FCIII+ cart (and with an original FC cart) there is no proble on the freeze menu on this machine.
4) What driver shall I install onto windows 10 in order ef3util.exe to work? The driver from the EF3 site cannot be installed as it is not signed.
5) What happens if I happen to turn off the machine while it is loading (flashing?) one of the .crt images - when I select those? Will the KFF be bricked? Or just the Action Replay will not start? Or?

Thanks a lot,
Sandor

dikdom, 2020-05-22:
Let's skip 3) as I have a slightly broken SID chip in this machine. With a different, c64c machine there is no problem with this cart. I have to check different SID also...

KimJorgensen, 2020-05-22:
Hi,
Good to hear that you got it working.

1. That depends on your last selection. Say you select BASIC (F7) then the cartridge starts in BASIC after reset or power on. If you don't wan't the cartridge to "remember" your BASIC selection this can be changed in the settings menu (F5)
2. Sorry, there isn't any longer description. Hopefully it is understandable what is does with my answer to your first question.
3. There are timing differences between different C64 machines and as the timing is close to the limit there is a chance that you encounter a machine where the cartridge doesn't work correctly. If you have the possibility to replace the SID chip let me know if the problem persists.
4. It shouldn't be necessary to install a driver as KFF is added as a COM port but you need the ef3usb.exe from the KungFuFlash GitHub page.
5. There is a slim chance that it will be bricked, but most likely the flash will be corrupted and KFF will start the menu next time it is powered on.

/Kim

dikdom, 2020-05-22:
Thanks!
1) so if I set the config in F5 to 'no' then it will always powers on onto the menu . If I set the config to 'yes' it will powers on into BASIC. The
resets will always takes you to the last selection (prg load, mount, etc..) regardless of the config. Did I get it right?
3) sure I will
4) hm.. I'm using the .exe from your github, so probably yes, it was ef3usb.exe . How long should it take to transfer one .prg file? Minutes? After 30s-60s I stopped the process as it seemed it crashed. Same with the testing. It prints it started but then nothing...

KimJorgensen, 2020-05-23:
1. No. Power up and reset works the same way.
In the default configuration KFF will always take you to the last selection which includes BASIC. If you set the config to 'no' then the BASIC selection wont get saved and KFF will go to whatever you selected prior to BASIC.
4. The transfer should only take a few seconds. What command exactly did you run, and did you make sure to specify the correct COM port?

dikdom, 2020-05-23:
point 1)
now I understand. KFF will always boots up to the latest selected action, the config only selects if these actions would include the BASIC screen or not. Ok, I was completely off the track with this.

point 3)
The working SID didn't solve the problem. In fact, when I started the freezer menu of FCIII on another 250407 board, then the menu line was also corrupt, but not that much, only within one raster line. Starting the KFF in a 425 board, then the freezer menu was OK. I have tested KFF on a 469 also, and there were no problems either.

point 4)
I executed 'mode' command to check the usable COM port. It was COM6 in my case. If I remember correctly it was 'ef3usb.exe com6 test' and 'ef3usb com6 c xy.prg' or something like that. The menu screen changed on the C64 screen, so something had happened, but not quite what I expected..
I will try different C64 machines. 

Good news is that I managed to test with the PLA20V8 PLA replacement and it works with it as with the original PLA.

KimJorgensen, 2020-05-23:
3) Thank you for testing this. I'll try to see if I can fix that once I got my 250407 board working, but it is quite tricky to get the timing right so that it works across all board revisions.
4) Just like EF3 you must send the ef3usb.prg program to the C64 first before you can use the test or copy command. So try to execute `ef3usb com6 s` first

Thank you for the info about the PLA. I'll update issue #12 with this.

dikdom, 2020-05-23:
4 - ef3usb.exe works as you described. I don't have an EF3 so I was not familiar with the procedure. And I stopped reading the doc at: 'PC has installed Easyflash 3 USB driver.' (quite the beginning of the readme.txt)
A small disappointment that this utility is not for transferring data to SD card, and quite a pleasant surprise that it is possible to transfer image/files onto real 1541 floppy disks.
Thanks for your support!

KimJorgensen, 2020-05-24:
Yes, the readme file should be updated.
The only change that was made to EasyFlash 3 USB Utilities was the how it communicates with the cartridge. No features were added to read or write to the SD card.

I don't think I would extend this utility with these features but rather have it build into the Launcher (see #13)

dikdom, 2020-05-24:
Ok, thanks!

closed 2020-05-24 by dikdom