KimJorgensen/KungFuFlash #18: Mount/disk change

dikdom, 2020-05-23:
Hello Kim,
- is there a way to switch D64 image during normal operation?
I managed to load Space Taxi (the trick is that you have to load the file with its name /LOADER.
But in case a two sided simple grame there is no way to play with it. Except one could switch the D64 without reseting the C64.

- is there a way to mount a D64 image **and** use a cartridge (AR or FC III or something else)?

Thanks!

KimJorgensen, 2020-05-23:
Hi!
No, currently there has not been implemented any functionality for changing the D64 image during normal operation. The only option for playing multi-sided D64 games is to copy all the files to a larger disk image (D71 or D81) if possible.
The disk emulation is quite limited and will never be able to fully replace a disk drive connected to the serial port. While it would be possible to implement support for a custom cartridge that can access the emulated disk drive, it wouldn't work with AR, FCIII or similar cartridges.

dikdom, 2020-05-23:
Ok, this is I saw, just I thought I doublecheck.

closed 2020-05-23 by dikdom