KimJorgensen/KungFuFlash #19: Lockups

lumenin, 2020-05-24:
I've been testing the cart for a few hours now, on a UK made C64 breadbin.  I don't have the board revision, but it's an early model I think.  The computer recently passed a diagnostic with test harness.

In normal use with the KFF, there are lockups every now and then.

1. Sometimes when powering on, the menu is displayed, but the list is empty.  Pressing F1 gives me the help screen, so the machine is not frozen.
2. Often, when selecting a d64 image to load using the * entry, it gives me a DEVICE NOT READY error.  But sometimes it succeeds.
3. When navigating the list in the menu, the machine locks up completely.  It seems to happen most often, but not always, when going into a subdirectory.

Now, the interesting part is that when I exchanged the SanDisk Ultra 16GB memory card for a Kingston Canvas Select Plus 16GB, all issues were much less frequent but not completely gone.  This *could* point to some FAT / card code related problem.  The one issue that was almost totally gone with the Kingston card, was d64 autostart.

With the Kingston card, there were still issues:

4. Jumpman Jr. crt game locks up after selecting number of players and player speed.  Always. Other carts, for example Moon Patrol, plays fine for hours. (I tested)
5. There is trouble launching prg files from time to time, so I may have to try several times before succeeding.

Moon Patrol being so stable may be an interesting clue here.  Also, does the KFF menu program behave like a static cart itself, or does it maybe challenge the timing from time to time?

I tried upgrading the firmware from 1.07 to 1.08 with the SanDisk card, and that made the issues much worse.  With the Kingston card, issues were maybe the same.

Regardless, I just love this project!  The idea of using the GPIO of a microcontroller emulate address-, data- and control-lines, and the functionality built in to the user interface - it's just awesome!

KimJorgensen, 2020-05-24:
Thank you for your observations.

I have developed KFF using a SanDisk Ultra 16GB card so it is weird that you see so many problems. Do you know if a replacement PLA or kernal has been installed in your C64? There are known problems with some of these.

It is interesting that the SD card would have such an effect on the stability, that is something I need to investigate further. Did the two cards have the same content or was the directory structure different?

Regarding Jumpman Jr. I can confirm that there is a problem with that game however I also get the same problem when I tried it with my EasyFlash 3 cartridge. I suspect there is a bug in the game as the "Jumpman Junior (1983)(Epyx)[a2].crt" version seems to work fine.

lumenin, 2020-05-24:
PLA is original.  Kernal was EPROM with switch between JiffyDOS and CBM.  All the above were with tested the CBM kernal chosen.

Btw, I was able to get into the first level of Jmpm Jr. crt, just after switching to JiffyDOS kernal.  I could not reproduce it, though, so it suggests that a) The game is OK, b) it is some kind of timing problem.  The crt image is the [a].crt, judging from the CRC-32.  The crt works fine in Vice 2.4.

To see if the EPROM was the problem, I put an original Commodore kernal in and tested carts, d64, prg for half an hour.  Issues were as before: Freeze in menu, no list in menu, Jmpm jr. no go (except the one case), and the occasional (but rare) DEVICE NOT READY.

Another interesting bit of information may be that the menu of EasyFlash crt-s have been dead stable.  I've not been wandering around in those VERY much, but - so far - no freezes there.

As for the lockups when trying to select an item or subdir in the menu - I tried to hit runstop-restore this time.  That produces the light blue border with dark blue screen combo with no text on screen.  From there, reset or menu is the only "option".

KimJorgensen, 2020-05-24:
Thank you for the additional information. It seems that all your problems are related to SD card access, besides the one with Jumpman Junior.

Was the content of the two SD cards identical? Has there been done any other modifications to the C64 board than replacing the kernal?

The reason for suspecting that there is a bug in Jumpman Junior is that the problem is very consistent, unlike the other problems you are seeing, and it also fails on my C64 which does not have any of the other problems that you describe. It could be that both EasyFlash 3 and KFF has the same bug that causes the game to fail, but the fact that there is an alternative version that does work could suggest that the bug was fixed at some point, maybe a NTSC/PAL issue? To know for sure we would need to debug the game or compare the disassembly of the two versions.

And yes. The only option to recover from a menu lockup is a restart (reset, menu or power cycle)

lumenin, 2020-05-24:
Yes, the contents of the SD cards were identical.

No other modifications, that I know of.  Well, except for the VIC-II chip having been put in a LumaFix socket.  All parts seem original.  PCB is the old type, with surplus solder on all the traces.  RAM is the old ceramic type.

Btw, sometimes the menu list froze with the letter 'b' close to the top left corner, other times with a column of 'b's all the way down.

I tried another crt, the Commodore Music Maker.  It is the only program started from the KFF that has had visual artefacts, namely black, flickering stripes on every 8 scanline or so.  This indicates some kind of bus collision.  If I remember correctly, all that the VIC-II chip sees, is RAM, or CHAR ROM, depending on banking settings.  I found a video of the program (apparently loaded from tape in the video): https://www.youtube.com/watch?v=ARvOiWEKcRQ

Jmjr: Could be, could also be that it does some memory banking gymnastics?  I don't know.  Btw, i did not try the jmjr from EF3, only KFF.

KimJorgensen, 2020-05-24:
I have never tested on such old C64 so it is quite possible that it has different timing requirements that the ones I have tested on.

I guess Music Machine is the Ultimax version. Is seems most of the Ultimax release have the VIC-II read data directly from the cartridge ROM instead of RAM. So this could indicate that the timing for the VIC-II cycle also is off for your C64. If you can find the 8kB Ultimax version of Wizard of Wor (not the 16kB C64 one) I would expect there to be similar artifacts. You could also try to active the freeze menu in FC-III which also runs in Ultimax mode.

lumenin, 2020-05-24:
Yes, I'm also starting to suspect the old board.  It is ASSY #250407, one of the oldest versions.

It was not Music Machine, it is called Music Maker aka Music Composer.

I'll check with the FC III Freezer menu + an Ultimax game to see what it reveals, and report back in a few days.

Maybe the C64C in the basement will work better?

lumenin, 2020-05-26:
Test results:

C64C: No lockups at all.  No artefacts in FCIII freeze menu.  No artefacts in WoW max.  No artefacts in Music Composer.  d64 mounting and loading of multipart game worked as intended.  Jmjr still no go.

Old board: Artefacts in FCIII freeze menu (menubar).  WoW max: Momentary flicker at all sprite locations when hero dies.

So here's an idea: Get an early revision board c64 to use as testbed.  If it works on that, it'll probably work on all models.

hackup, 2020-05-26:
I just tried some of the mentioned issues an can confirm the following:

- Slight artifacts on the FC3 freeze menu bar on a 250407, no artifacts on a 250425.
- Heavy artifacts in WoW ultimax both on a 250407 and a 250425. The same CRT runs fine on both boards when using an EF3.
- Jumpman Jr. crashes on all boards, both when using the KFF and the EF3. So the CRT does seem to be buggy.

lumenin, 2020-05-27:
It is interesting, though, that the same Jmjr .crt works in Vice, and that I got it to run once on the old C64 with KFF.  I wonder what happens if it is burnt to an EPROM and run from a real cartridge... :-)

KimJorgensen, 2020-05-27:
@hackup Thanks for testing this which confirms that the timing for the VIC-II cycle is a bit off.

@lumenin I have compared the disassembly for the two versions, and there is only a small difference.

Jumpman Junior (1983)(Epyx)[a].crt (craches):
```
.C:9974  18          CLC
.C:9975  A5 B2       LDA $B2
.C:9977  65 B3       ADC $B3
.C:9979  C9 3F       CMP #$3F
.C:997b  D0 04       BNE $9981
.C:997d  78          SEI
.C:997e  8D 16 D0    STA $D016
.C:9981  20 22 99    JSR $9922
```
Jumpman Junior (1983)(Epyx)[a2].crt (works):
```
.C:9974  18          CLC
.C:9975  A5 B2       LDA $B2
.C:9977  65 B3       ADC $B3
.C:9979  C9 3F       CMP #$3F
.C:997b  D0 04       BNE $9981
.C:997d  EA          NOP
.C:997e  EA          NOP
.C:997f  EA          NOP
.C:9980  EA          NOP
.C:9981  20 22 99    JSR $9922
```
So there has clearly been made a patch in the [a2] version which seems to disable some kind of copy protection. The game checks the C64 has already been running prior to the game by checking if the pointer to datasette buffer has the default value.

To make it work on KFF: Select the game to have it auto start. Power off the C64 until the content of the RAM is erased (~10 seconds). Power on the C64 again and the game should work.

Now that I know this, I could have KFF erase that memory when starting a cartridge from the menu.

dikdom, 2020-05-28:
Regarding to the original lockup in menu issue...
I have a 407 board and a KFF cart. I go to the menu, browse the SD card. In one of the directories there are 250+ files. I just go down and up (using page scroll) till I hit top and bottom and top, etc... again and again. 
On the 407 board the browser locks up after about 5-10 top-bottom move. It is not consistent, not always at the same position..
On a 469 I could find this issue. I repeated more than 10 times and it just worked.

KimJorgensen, 2020-06-15:
Could you please retest with version 1.09 which should address these lockups

dikdom, 2020-06-18:
For me, it works. For some strange reason I could hardly reproduce the dir-lockup issue on my 407 board with v1.08, but finally I managed. With v1.09 I couldn't make KFF to hang in that menu, even minutes I moved in the menu up and down..
And the FC-III freeze menu is also got fixed!! Thanks a lot!

lumenin, 2020-06-28:
Tested 1.09 now, here are my results:

Freezes / strange characters in KFF menu are now gone. Wonderful! :-) With 1.08 the menu locked up more often than not.

Jmjr: No change from 1.08, so I did some more testing.  Conclusion:  Game does not work when started from menu. Hardware reset switch does not fix it, neither KFF reset switch.  But it works after power cycle!  Then I can do both hardware and KFF reset, and the game starts every time.

FC III freeze menu, WoW max and Music Composer all have graphics artefacts, as before.  No change.  WoW looked a tiny bit worse, even.

lumenin, 2020-06-28:
Btw, for the artefacts in Music Composer, I had one situation where I got a lot of artefacts, then did some kind of reset / power cycle / start from menu again (don't remember, sorry).  The level of artefacts were down to one sixth or something, just like that.  Hope this helps.

KimJorgensen, 2020-06-28:
Thanks for testing.
There hasn't been made any changes in 1.09 that adresses Jumpman Junior but I'll include one in the next version. For now you need to power off the C64 to circumvent the copy protection.

lumenin, 2020-06-28:
I thought this one was about jmjr, was it not part of 1.09 binary release?
 
https://github.com/KimJorgensen/KungFuFlash/commit/3a577d11270a7c59572c34b27a149d18eaddd0fd

KimJorgensen, 2020-06-28:
No, I made that after the 1.09 release.

KimJorgensen, 2020-07-31:
The fix for Jumpman Junior has been included in the 1.10 release so with that all the lockup issues should be resolved.
Remaining are the graphics artefacts in the FC III freeze menu, WoW max and Music Composer

KimJorgensen, 2021-04-11:
I'll close this as all the original issues has been resolved

closed 2021-04-11 by KimJorgensen