KimJorgensen/KungFuFlash #20: Feature request: Mount .d64 with tool crt active

lumenin, 2020-05-24:
So, for example, having a .d64 mounted with Simon's Basic active.

Somewhat (un)related: Be able to mount .d64 with the tapecart-SD browser.  I could not find an issue list for it so I mention it here.

KimJorgensen, 2020-05-24:
It is a nice and understandable request, but unlikely to get implemented on the current hardware platform.
It would be possible to implement support for a custom cartridge that can access the emulated disk drive. However, it wouldn't be a general solution as this would require an extension of every supported cartridge type if possible and many of the .crt files would have to be patched, dependent on how they access the disk drive.

lumenin, 2020-05-26:
This could be done in several iterations:

1) Trivial - keep tool cart (ccs mon, simons basic, or similar) active, while mounting the d64 in the usual way by modifying the I/O vectors (or what you do) to redirect device 8 loading to KFF.  
2) More work: Cart patching
3) Hehe, even more: d64 write support.

As for 2), it will probably make more sense to replace fastloading routines with KFF loading in modified versions of the carts.  So, no patching per se, but new cart versions.  This, of course, will make them useless against real hardware, unless there is room for both fastload and KFF routines in the cart.  This idea does not depend on any development in the KFF itself, so it could be done separately.

Gabeki17, 2023-02-24:
Hi,

I have done this different in the end.
I was tobtry with Epyx fastload. So that is resident.

I have edited the source of Epyx fastload and have added a simple load command. So when it starts, Epyx fastload activates and loads a prg for me now. 
And not even in a complicated way. It is using keyboard buffer simpley.

So no exra resouece is needed from KFF hardware or such.

So the onley thing kff could do is to fill up keyboard buffer, so it executes. 
Not very elegant, but effective.

With regards, 
Gábor

closed 2024-12-29 by KimJorgensen