KimJorgensen/KungFuFlash #21: PCB rev history is not available

dikdom, 2020-05-25:
Hello,
You made the rev2 gerber files available, but in the meantime the rev1 files are vanished from the release history.

I created the rev1 SMT production files for JLCPCB and wanted to share. They are tested. I know it is obsolete, so let me know if you would like to publish.

KimJorgensen, 2020-05-25:
The rev. 1 Gerber files has not been removed and can be found here: https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.00
Yes, please feel free to share the SMT production files and I'll update the release with these

dikdom, 2020-05-25:
My bad. I just checked v1.01 and of course the gerbers were not there. Sorry for that...
Here are the manufacture files for JLCPCB. USB and SD card sockets and tack buttons are not there as I couldn't find those at them. I guess they are not offering these at the moment.

[KungFuFlash-rev1-SMT-JLCPCB.zip](https://github.com/KimJorgensen/KungFuFlash/files/4676689/KungFuFlash-rev1-SMT-JLCPCB.zip)

For the USB micro B socket I used: Amphenol, 10118194-0001LF.
For the micro SD socket I used a push-push socket: Attend, 112J-TDAR-R01
But these were ordered separately and I soldered them manually.

KimJorgensen, 2020-05-25:
Thank you for your contribution.
I have added the zip file to the release and updated the BOM with the part numbers.

closed 2020-05-25 by KimJorgensen