KimJorgensen/KungFuFlash #22: Unstable EF3USB

dikdom, 2020-05-25:
Using win10 and ef3usb.exe I managed to connect with KFF and transfer D64 images to floppy disk. Most of the time. But sometimes it just hangs up.

Below you can find the output of the ef3usb.exe of one hang. The transfer stopped at T:4 S:4. The floppy got silent, the C64 screen didn't move, stopped at Track 4, Sector 4. 
With a MENU it went straight to the Menu (most of the time. Sometime I had to press menu 2-3 times for that).

By the way, one D64 image transfer without verification took around 2m10s. With verify it was 4:30.

If you'd need further information, just let me know what to check!

```

c:\xzy>ef3usb.exe com6 s
EF3 USB Client v1.93 - Kung Fu Flash version

 - SEND PRG TO EF3 MENU

Handshake: EFSTART:PRG
Response: [LOAD]

 - Sending PRG file. Bytes sent:  13080

DONE!

c:\xzy>ef3usb.exe com6 w "Creatures_a.d64"
EF3 USB Client v1.93 - Kung Fu Flash version

 - WRITE IMAGE

 - .D64 format 35 tracks.

Transferring image to C64. ^C 4 S: 4
c:\xzy>ef3usb.exe com6 s
EF3 USB Client v1.93 - Kung Fu Flash version

 - SEND PRG TO EF3 MENU

Handshake: EFSTART:PRG
Response: [LOAD]

 - Sending PRG file. Bytes sent:  13080

DONE!

c:\xzy>ef3usb.exe com6 w "Creatures_a.d64"
EF3 USB Client v1.93 - Kung Fu Flash version

 - WRITE IMAGE

 - .D64 format 35 tracks.

Transferring image to C64. T:35 S: 0
Done

c:\xzy>ef3usb.exe com6 w "Creatures_b.d64"
EF3 USB Client v1.93 - Kung Fu Flash version

 - WRITE IMAGE

 - .D64 format 35 tracks.

Transferring image to C64. T:35 S: 0
Done

```

KimJorgensen, 2020-06-17:
Could you please retest with version 1.09 to see if that fixes the problem?

dikdom, 2020-06-23:
I did and unfortunately it didn't disappear. I would say it happens more seldomly but it happens.

dikdom, 2020-06-24:
Could you please guide me how to compile that two files (ef3usb.c and ef3usb.tas). Otherwise...
Could you build for me a debug version of these utilities? We shall somehow figure out where does the communication break.. Pressing RUN/STOP + RESTORE on C64 does what is expected, so the computer doesn't hang completely..
- ef3usb.c
  In d64 writer subroutine (command==2) could you please check every written byte count with the expected numbers (and if it fails, printing the error point, relevant values and breaking from the loop) in the turbo branch (line 852-909)
- ef3usb.tas
  Maybe the read_byte routine could contain some counter and break if there are too much wait happened.

But if you are already there. Could you please move with 3 lines up the 
```
ldy #$00
lda #$17      ; 'W'
sta (indaddlo),y
```
code in the write image kernal (1226-1229) to write 'W' first, the write that track. And writing a 'V' for verify also could be a nice to have feature.
And, if you already there. Could you change the sectors read/write order instead of being linear to (0 3 6 9 12 15 18 21 1 4 7 etc...), to have a little time for spin the disk.. This may need some measurement..
And if you send those compiled files (please provide the sources also, to ease the debugging) then I could tests those...

KimJorgensen, 2020-06-25:
There is nothing special needed for compiling ef3usb.c, so for Linux `gcc -O2 ef3usb.c -o ef3usb` and for Windows I think it was `cl /O2 ef3usb.c` in a Visual Studio Command Prompt.
I didn't make any changes to the C64 source or binary so I don't any instructions or other source files than provided in the original [EasyFlash 3 USB Utilities release](https://csdb.dk/release/?id=150097)

dikdom, 2020-06-30:
Thanks, I will try this.

dikdom, 2020-07-03:
Thanks!
I already manage to build the PC side program (.c) for linux, but I failed to do so with the .tas files. There are missing include files. Is it supposed to be built by 64tass?

KimJorgensen, 2020-07-03:
It could look like it is supposed to be built by 64tass but as I haven't built it myself I can't really help you with this.

rickard-von-essen, 2020-07-05:
According to Tom-Cat 

> > Nice one, but sadly the Archive doesnt contain all sourcefiles to compile the C64 Version, could you please fix that ? Or is there anything that is ment not to be spread ?

> Yes, unfortunetly I can't spread the file browser source, you can get the tlr Format II sources needed from its release and generate the turbo write/read functions with the Plushdos (2bit irq loader on 2000 and 2500 addresses) - those are not mine to spread either.

https://www.forum64.de/index.php?thread/47071-ef3-usb-utilities-1-6-with-sources/&postID=590319#post590319

KimJorgensen, 2020-07-05:
Tom-Cat did provide one of the missing source files https://www.forum64.de/index.php?thread/96247-kung-fu-flash-cartridge/&postID=1540367#post1540367 and the rest could be found here https://svn.icomp.de/svn/chaco/chusb/

I have added these files to repository and provided a Makefile so it should now be possible to build the C64 program

dikdom, 2020-07-06:
Thanks for the investigation and the files! :)

dikdom, 2020-07-07:
erm... what version of 64tass do you use? Probably some changes in the 64tass broke the backward compa...
I downloaded and compiled the latest,
`64tass Turbo Assembler Macro V1.55.2200`
but it failed to make the 3rd party. It complained about the .proc didn't have any label. Like this 5-7 times: 
```
taper.tas:29:1: error: label required
 .proc
 ^
```
When I moved the .proc statement into the same line as its label (usually in prev line) the make executed with two warnings and I got the binary for C64 also.
The linux native ef3usb compiles without any hiccup.

Managed to speed up the disk image turbo transfer from 2:08 to 1:37 (~25%), but it is still hangs during data transfer between PC and KFF card.
[ef3usb-kff.zip](https://github.com/KimJorgensen/KungFuFlash/files/4887166/ef3usb-kff.zip)
If you are allowed to modify the sources I'm happy to provide those as a pull request.

Unfortunately the transfer hangs on a 469 mobo C64 also..

KimJorgensen, 2020-07-08:
By trial and error I found a version of 64tass that did work and documented that in the Makefile. I did also try the same thing with the .proc statements and V1.55.2200 but the resulting binary was not identical to the ef3usb.prg file from the original EasyFlash 3 USB Utilities release. It could be that these differences are insignificant, but I haven't really investigated that.

I would be happy to accept a pull request and don't think that there is a problem modifying the source as long as we update the header to state that this is a modified version. If the original author disagrees I would of course respect that.

dikdom, 2020-07-10:
I didn't notice the comment in the Makefile, that's why I had problems. Anyway, I fxed the source for the latest compiler.
Pull request created.

KimJorgensen, 2020-07-12:
The PR has been merged now. I changed the taper.tas file a bit otherwise the code in `end_of_transfer` wouldn't get included. The ef3usb.c was also changed a bit to make it build on Windows.

KimJorgensen, 2021-08-12:
The original issue, unstable EF3USB, has been fixed in release v1.24.

closed 2021-08-12 by KimJorgensen