KimJorgensen/KungFuFlash #23: Sorted directory list

dikdom, 2020-06-12:
Hello! Would it be possible somehow to order the directory list alphabetically? And always put the directories in the beginning of the list?
Currently it is listed as one wrote to the SD card. Sometimes it is hard to find a title.

KimJorgensen, 2020-06-13:
No sorting has been implemented in KFF as this would make the navigation really slow for large directories. However, CapFuture1975 on forum64 suggested that you can use a tool on the PC to sort the files on the SD card.
http://www.anerty.net/software/file/DriveSort/

dikdom, 2020-06-17:
What is the STM32 side retrieves the full directory (if it fits into its memory) and sorts it.. That could be processed fast.

KimJorgensen, 2020-06-17:
Large directories won't fit in the available memory and that is why sorting would be really slow.

dikdom, 2020-06-30:
Ok, thanks!

closed 2020-06-30 by dikdom

jmredfern, 2023-04-20:
@KimJorgensen what about the STM uses a paging file as virtual ram on the sdcard to do the sorting? just a thought, thanks for the cart!