KimJorgensen/KungFuFlash #28: Question

dikdom, 2020-07-20:
Hello Kim,

this is not an issue, just a question... I would like to know what was the reason you left out the file ef3usb.tas from the https://github.com/KimJorgensen/KungFuFlash/pull/26 pull request? That file contained non-functional modifications but would have helped depug the locking issue to locate (by printing 'W' and 'V' in proper timing).

My second question: would you allow me to implement the 'SAVE' functionality on a mounted D64 image? Or do you already work on it? (not started yet, but spent some time investigating on it - definitely I would have some time to implement it right)

KimJorgensen, 2020-07-20:
Hi dikdom,
Not sure why you think that ef3usb.tas was left out? It looks merged to me https://github.com/KimJorgensen/KungFuFlash/commit/3ed188195505601571c900d6aba7a95d607aeb27#diff-af5515dba1e75e95657cd5c581d91a8a

Yes, for the D64 save functionality we need to optimize the EF RAM usage first

dikdom, 2020-07-20:
Er... probably I was confused by the timestamps. The ef3usb.tas file has an older timestamp, but indeed. The 'W' is in the right place... Sorry.

EF RAM usage on trampoline codes...
I was wondering if the filename could be copied only if it is really needed (from under $8000-$BFFF), and if needed then let's use the datasette buffer ($033C-$03FB). I know that increases the risk of being non-compatible with some programs, but I hope, not much.. Or, the lower part of the stack also could be used for file name storage also.

dikdom, 2020-07-22:
@KimJorgensen I have an idea but that is kind of a hack..
So, currently the RAM area is initialized statically: at the point of exit-to-BASIC the EF RAM content is filled with the vectors and code and it stays there until RESET (ok, filename, and some variables are overwritten time to time).
What I would change is that at the exit-to-BASIC the EF RAM content is initialized with the vectors (including new ones that currently doesn't fit into the $FF limit) and only with the 'store_filename_enable_ef' method (including the 'enable_ef_rom'). The rest would be dynamically copied there when needed. Load, and open requires the 'kernal_trampoline' (and load_prg also for the kff_load). 
And when the load/open/etc is finished, it shall return to EF ROM and copy there the 'disable_ef_rom' routine at the end of the 'enable_ef_rom' (this must remain there!) and thus the C64 can leave safely the EF ROM routine. But it requires some extra copy routine and very conscious layout of the RAM routines...
What do you think?

KimJorgensen, 2020-07-22:
@dikdom I was hoping that it would be possible to optimize the current code or move more of it to ROM to make room for the SAVE functionality but I haven't spend enough time on it to know if that is feasible.

dikdom, 2020-07-22:
I'm sure some bytes could be saved on optimizing the code but surely not enough to squeeze in a new method's helper functions. Not to mention three new methods' (save, chkout, chrout).
I'll try to refactor and after to start implementing the save function. I would like to see if it's possible at all.

dikdom, 2020-07-23:
@KimJorgensen : I'm afraid I made a terrible mistake and I'm really sorry, just please help me to fix it... 
I forked your repo and in the forked repo I created a new branch called 'save'. I started the refactor and I think I managed to refactor the disk.s file to have space in the EF RAM for the save and related routines (even a longer, 38 char long filename). I pushed this branch to my forked repo ('git push origin save'). But now I see this branch on your upstream repo too... How on Earth this happened??

KimJorgensen, 2020-07-30:
@dikdom Please note that the current implementation requires you to disable the C64 interface while writing to the SD card. See https://github.com/KimJorgensen/KungFuFlash/blob/master/firmware/stm32f4xx/diskio.c#L554
This was done to avoid TX FIFO underruns

dikdom, 2020-08-02:
Thanks, I will try this as from the merge from v1.10 the d64 images don't change. I have some other issues too, some testing, hopefully by the end of the next week I'll be able to present the SAVE command to you...

dikdom, 2020-08-02:
@KimJorgensen: I'm stucked now and I think there is something I should know but I don't...
- is it OK from the C64 side from the disk.s to call the `ef3usb_send_byte` followed immediately by a `ef3usb_receive_byte`? I
mean, the MCU side realizes that there is data in EF_ROM, it processes, then responses some result (presumably OK). Can you see any timing issue here? (the MCU may or may not write to SD between send and receive calls) 
- on the MCU side I'd need a bigger buffer (~800bytes) and I figured out that I could use the `crt_ram_banks[1]` area, right after the channels this way:
```
D64_SAVE_BUFFER *save_buffer = 
        (D64_SAVE_BUFFER *) (((uint8_t *)crt_ram_banks[1]) + sizeof(DISK_CHANNEL)*16);
```
Do you think it is feasible (there are 16 channels used) or am I interfering with some other method?
I even tried to use the `scratch_buf `instead of the `crt_ram_banks[1]` but no luck.
```
D64_SAVE_BUFFER *save_buffer = 
        (D64_SAVE_BUFFER *) scratch_buf;
```
- same approach different problem. I thought I could make the C64 and the MCU communicate with each other using the EF_RAM area. 
```
uint8_t *receive_buffer = (uint8_t *)&crt_ram_buf[RECV_BUFFER_OFFSET];
```
In fact, the upper side of the ef_ram (0x90 bytes currently, but this my change) is used to transfer the actual data, controlling the transfer is done by ef3usb_transfer and _receive calls.

Currently the D64 file is written (*), but the C64 crashes. I have no idea why. It seems that the last `ef3usb_receive_byte` doesn't return, just crash but I can see no reason for that. That code is in the ROM of the C64, so probably there is some memory corruption on the MCU side but I simply can't find the reason...

(*) - when a single sector program is saved, currently the program is properly written into the D64 but the C64 crashes. In case a 2 sector program is saved, the first sector is saved, the second sector is written to a completely wrong position (0x0000ff00) in the file, but the first sector link is correct (and the C64 crashes), after that the BAM and the directory is updated (from save_buffer). I'm pretty much clueless..
Any idea would be appreciated...

KimJorgensen, 2020-08-03:
> is it OK from the C64 side from the disk.s to call the ef3usb_send_byte followed immediately by a ef3usb_receive_byte?

In general there is no problem in doing that and that is already done in the existing code.

> Can you see any timing issue here? (the MCU may or may not write to SD between send and receive calls)

As long as the MCU is ready to receive the byte there shouldn't be a timing issue. However, as I wrote earlier, you need to disable the C64 interface if you want to write to the SD card and while the interface is disabled the EF ROM, RAM, and registers aren't available on the C64 side. So special care needs to be taken if you want to disable the C64 interface while the C64 side is waiting for a response. For an example on how to do that, see the implementation of ef3usb_get_cmd or REPLY_WRITE_WAIT.

> Do you think it is feasible (there are 16 channels used) or am I interfering with some other method?

That should be OK. If you need a buffer I would create a new struct that contains the 16 channels and the new buffer. That way you don't need to manually calculate the memory offset.
For what purpose do you need an extra buffer? Can't the existing DISK_CHANNEL be used?

> Currently the D64 file is written (*), but the C64 crashes. I have no idea why.

Did you disable the C64 interface on the MCU side by calling `c64_interface(false)` before trying to write to the SD card?

dikdom, 2020-08-03:
> and while the interface is disabled the EF ROM, RAM, and registers aren't available on the C64 side

This is the key - again my understanding was the problem. It was not that obvious to me that closing the C64 interface means that there is no code shall be running on C64 from the EF ROM/RAM (what else could it be?? sorry...)
ef3usb_get_cmd() is compiled into the .data section (0x0800), I guess the normal startup copies this code into the proper area. Similarly I guess the eapi-kung_fu_flash.s is compiled and copied to $F800, in C64 RAM, under the KERNAL ROM. Is it the case?
So basically one has to wait in C64 RAM area and wait for a well known return string - in case the c64 interface was turned off in the meantime.

Answering your questions:
> For what purpose do you need an extra buffer? Can't the existing DISK_CHANNEL be used?
- I need to change the BAM and directory before writing the data onto the disk, but I want to write these changes when all the data is already written into the D64 file. So I keep those changes and when the save is finished I write to disk the BAM+dir sector(s). I could be a bit more space conscious, but that would be a second iteration.
- Existing DISK_CHANNEL shall not be used as there could be open channels during program save.

> Did you disable the C64 interface on the MCU 
- No, I did not, but as I call the write_file() (implemented in _filesystem.c_) and it calls f_write() (implemented in _ff.c_) and it calls to disk_write() (implemented in _diskio.c_) and as I understand that last one manages the c64_interface by calling __disable_irq() / __enable_irq().
I believe this is enough, right?

KimJorgensen, 2020-08-03:
> ef3usb_get_cmd() is compiled into the .data section (0x0800), I guess the normal startup copies this code into the proper area.

Yes, this is done in crt0.crt.s by calling copydata.

> Similarly I guess the eapi-kung_fu_flash.s is compiled and copied to $F800, in C64 RAM, under the KERNAL ROM. Is it the case?

No, as you can see in ld.crt.cfg there is not specified a `run` attribute and it is not copied to RAM by the startup code. The code can be accessed at $f800 in Ultimax mode (or at $b800 in "16k cartridge mode"). The EAPI code is required to be relocatable and it is only the code and data in the EF RAM that has a fixed location.

> So basically one has to wait in C64 RAM area and wait for a well known return string - in case the c64 interface was turned off in the meantime.

Yes, this was the method I used for KFF. When the interface is disabled you basically read random data on the bus, so you can't rely on a single register value.
Another solution would be to trigger an interrupt on the C64 using the IRQ line.

> Existing DISK_CHANNEL shall not be used as there could be open channels during program save.

OK. I guess you are only looking at implementing the ISAVE vector (not IOPEN + ICKOUT)

>  ... as I understand that last one manages the c64_interface by calling __disable_irq() / __enable_irq().
I believe this is enough, right?

I don't think so. The interrupt flag will still be set causing the interrupt handler to trigger immediately when `__enable_irq()` is called which could violate the C64 bus timing. At least that is what I think will happen, I haven't tested it.

dikdom, 2020-08-03:
@KimJorgensen 
It works, thanks to you! I can save (and load back) 2-3 blocks basic programs. 123 block save fails, bughunting continues.. :)

dikdom, 2020-08-03:
> No, as you can see in ld.crt.cfg there is not specified a run attribute and it is not copied to RAM by the startup code. The code can be accessed at $f800 in Ultimax mode (or at $b800 in "16k [...]

Hm. `writeWait`, `waitDone`, etc. are implemented in eapi-kung_fu_flash.s, so those methods will be compiled into $F800 and will be executed from the EF_ROM in ULTIMAX mode? That can't be right. You say 'relocatable'. Who will copy the eapi code to a different (preferably c64 RAM) location from $F800 (in Ultimax mode)?

>  guess you are only looking at implementing the ISAVE vector

Yes, this is the first iteration (I wish it would run). The remainings (open for write, print#, D71/D81 support, IO commands like Scratch, New, Copy) are iteration n+1. 

> I don't think so. The interrupt flag [...]

You are probably right, I also inserted immediately the `c64_interface()` calls. I didn't dare to try without them (on a not-yet-working code)

129 blocks are saved in around 5-10 secs. For the first time (and now it can be read back!!). The second time it fails, so there must be some garbage left somewhere. This needs to be found and fixed and the overwrite (freeing up the file to be overwritten) also needs to be implemented.

BTW. When I finished with it can I send a pull-request?

KimJorgensen, 2020-08-04:
> Hm. writeWait, waitDone, etc. are implemented in eapi-kung_fu_flash.s, so those methods will be compiled into $F800 and will be executed from the EF_ROM in ULTIMAX mode? That can't be right. 

I didn't say that the code will be executed from ROM. It is placed at this well known location ($f800) so that it can be detected and copied.

> Who will copy the eapi code to a different (preferably c64 RAM) location from $F800 (in Ultimax mode)?

The program that uses the EAPI must copy the code to C64 RAM. You can read more about this [here](http://skoe.de/easyflash/files/devdocs/EasyFlash-ProgRef.pdf#page=11).

> When I finished with it can I send a pull-request?
Sure. I guess this will be a large PR so just be prepared for a some comments :)

dikdom, 2020-08-04:
EAPI code execution, doc: that explains the location of the code. Thanks!
Sure, it is a large PR, so I expect a lots of comments. Next week I will be on holiday. I hope I manage to fix/clean up the code and provide the PR this week.

dikdom, 2020-08-04:
Finally everything seems ok...
The pseudo code on the C64 side that makes sure that the MCU turned back its c64 interface (MCU sends 'OK' to notify c64):
```
a: wait for a character
  if it is not 'o', go to a:
b: wait for a character
  if it is a 'k', go to 'c'
  if is is an 'o' go to 'b'  ; this was a tricky one, took some time to analyze the random freeze...
  otherwise go to 'a'
c: <c64 interface is open, c64 can safely receive another code>
```

KimJorgensen, 2020-08-04:
I used a longer string to reduce the risk of erroneously reading the awaited string from the random data on the bus.
However, I should probably change the code to also check for the first character if the check for a subsequent character fails.

dikdom, 2020-08-05:
I found 2 letters just fine. Without the extra check the save failed by the second but latest by the third try. With the extra check I didn't meet any save failure (tried 16+ times consequently). 
Pull-request sent, have fun! :) 
(btw: I have found a minor .rel handling bug, I will reveal it when my PL is merged! :). )

KimJorgensen, 2020-08-05:
I would expect this to be hardware dependent so just because it works on your C64 doesn't necessarily mean it would work for everybody else.
I'll have a look at the PR tomorrow.

dikdom, 2020-08-07:
Having more than 2 different characters in the sequence would make the sequence a bit more complicated. Let's wait 'abc' :
```
A: wait for character
   if it is not 'a', go to A:
B: wait for character
  if it is an 'a', go to B:  // the unfortunate case when 'a' was on the floating 
                             // DATA lines and we cannot differentiate it from the on-purpose 'a' on the DATA lines.
  it it is not a 'b' go to A:
C: wait for a character
  if it is 'a' go to B:   // the unlike case when 'a' then 'b' and again 'a' were on the floating DATA lines (last 'a' can be on purpose)
  if it is not 'c' go to A:
D: wait for a character
// this is the char we are getting after the interface is open
```
This way at each character test step in the sequence you have to check if it is the first character, because then you have to jump to the second char test. I don't think that the floating data lines can float between exactly those two selected characters - by coincidence - that's why I think two characters are enough. But I agree that having more than 2 characters would lower the possibility of mistake.
I used two characters as these tests has to be in C64 RAM and during SAVE I don't have many areas I could use. So I selected the lower part of the CPU stack (used by datasette and FP transformation also) and I wanted to minimize the RAM usage.

KimJorgensen, 2020-08-07:
Did you see how this is implemented in the EAPI driver https://github.com/KimJorgensen/KungFuFlash/blob/master/launcher/eapi-kung_fu_flash.s#L455. It is quite small and simple.

BTW I have started looking at your PR but as it is quite large and I only have limited time at the moment it may take a while before I'm done.

dikdom, 2020-08-07:
Your code is good to read. I'll be patiently waiting for your comments!