KimJorgensen/KungFuFlash #29: Blocks free

dikdom, 2020-07-29:
I've checked it with 4 different d64 images on my SD card.
With the the v1.08 firmware I mount a D64, load a directory and the LIST prints a valid number before the 'blocks free'.
With v1.09 it prints 15922 blocks free for every disk. Going back to v1.08 it prints a normal value again.

With a 407 and a 469 motherboard also.

KimJorgensen, 2020-07-30:
Thank for reporting this.
It seems that this bug was caused by upgrading to a newer version of the GCC compiler. I have rewritten the code that caused this and the fix will be included in the next release.

dikdom, 2020-07-30:
Ok.
In the meantime I'm almost finished with the SAVE, I have only some minor issues... (well... not really.. I'm struggling to return to the basic READY. prompt properly, and for some reason I cannot write to the SD card though there is no error responses on the f_write operations).
So I hope the change you mentioned above is not a big one...

KimJorgensen, 2020-07-30:
This has been fixed in release 1.10.
It was a small bug fix but the EAPI support in that release was quite a big one

closed 2020-07-30 by KimJorgensen