KimJorgensen/KungFuFlash #33: Loading D64s issue (with JiffyDOS)

drakonia667, 2020-08-27:
I have a strange issue - when i want to run a file from D64 image, i have always "DEVICE NOT PRESENT ERROR".
PAL machine, but (maybe it causes that) JiffyDOS is installed inside.
I've tried three different microSD cards, same issue.
CRTs & PRGs works fine.
Can anybody help me with that?

dikdom, 2020-08-28:
Can you disable JiffyDOS and try to load that way?

KimJorgensen, 2020-08-29:
The disk drive emulator has only been developed and tested using the stock C64 kernal.
If you don't have the option of switching kernals then you should still be able to load single file games from a D64 image. Just select the file in the launcher and open the option menu (shift+return or long press fire) and select "Load" instead of the default "Mount and load".

drakonia667, 2020-08-30:
Yeah, it worked. I've missed information that KungFu doesn't support fastloaders.
Thanks!

KimJorgensen, 2020-12-27:
This issue should be fixed in release 1.19. Please let me know if it works for you

KimJorgensen, 2021-02-06:
As I haven't received any feedback I'll assume that this issue has been fixed. Feel free to open a new issue if this is not the case

closed 2021-02-06 by KimJorgensen