KimJorgensen/KungFuFlash #34: DFU Mode

Icelvlan88, 2020-09-18:
Hello,

I cannot get my unit to go into DFU mode. I checked all my connections and I just can't get it to go. I shorted J1 and BOOT0 to 3.3v.

I also checked continuity between my usb pins and the processor. I'm not getting any voltage. I also tried using a different USB cable and no go. I tried both applications you suggested swell.

Can you give me any ideas on what I should be trying? Is there something that is supposed to be sitting between the usb cable and the PC?

C-64Bram, 2020-09-18:
I had the same problem. After rebooting my pc the card became visible in DFU mode. It the device is removed and reconnected ,  it sometimes becomes visible again and sometimes I have to reboot again.

jgilcas, 2020-09-18:
Same issue here, i bought a stvlink programmer and everything works perfectly, for 2e i dont expend time to solder jumpers pins

C-64Bram, 2020-09-18:
Like this?

https://www.st.com/en/development-tools/st-link-v2.html

jgilcas, 2020-09-18:
This one: https://es.aliexpress.com/item/32719963657.html?spm=a2g0s.9042311.0.0.274263c07ACJ8a

C-64Bram, 2020-09-18:
Nice. :-)
Just ordered one. :-)

Icelvlan88, 2020-09-18:
I ordered one from a different vendor and receive a STC Auto Programmer USB-TTL. Will this function? Pins looks to have different names

dikdom, 2020-09-24:
Hello,
check, if you have 5V between the J1 pin header (if it is shorted then any of its pins) and ground (shield of USB). If you have 5V check the voltage regulator whether you have the 3.3V at its output. If you have it, you should check the MCU also to have the 3.3V . Make sure the two pins of J1 and 3.3V and BOOT0 pins on J4 are shorted.
After connected everything, press the reset button of the card, it helped to be identifiable by my PC.
I used dfu-utils and the latest .upd binary to flash my card. It worked for me.

closed 2024-12-29 by KimJorgensen