KimJorgensen/KungFuFlash #35: Does not work on my C-64

C-64Bram, 2020-09-18:
hi! Build myself the Kung Fu Flash.  Great cartridge!! 

As described in Github I booted the device in DFU mode.
Renamed the firmware from upd to bin and used DFU file manager to create a dfu file.
Flashed the file with DefuSe and verified. No errors. So far everything ok. Removed the 2 jumers and went to my C-64.
When I put the cartridge in the C-64, the screen stays black and the cart won"t start.
I've tried different firmwares but no result.

jgilcas, 2020-09-18:
Could be a NTSC C64? did you check 100R array resistors?

C-64Bram, 2020-09-18:
It's pal. Checked the array resistors and they are in place.

jgilcas, 2020-09-18:
Could you try to use dfu-utils? there is a win64 version available, working fine with following command dfu-util -a 0 -s 0x08000000 -D KungFuFlash_v1.10.upd

C-64Bram, 2020-09-18:
Is there a 32 bit version? I have 32 bit windows  :-)

jgilcas, 2020-09-18:
https://sourceforge.net/projects/dfu-util/files/dfu-util-0.8-binaries/win32-mingw32/

C-64Bram, 2020-09-18:
Thanks! Will try it later. Have to go to work now.... :-(

C-64Bram, 2020-09-19:
Tried with DFU-Utils. Did not work. :-( 
I quess because of the suffix because what the error sais... :-))

**Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Cannot open DFU device 0483:df11
No DFU capable USB device available****

So I used DFU file manager to create a DFU file from the BIN -file. Strangely the created DFU file is only 336 bytes. So something is going wrong.......

dikdom, 2020-09-24:
Hello,
with the dfu-utils I used the .upd binary what I have downloaded from here (github), without any conversation (I used linux under VirtualBox under win10).

C-64Bram, 2020-09-24:
hi. if I load the .upd-file with DFU-utils it says "this file does not have the correct format".

dikdom, 2020-10-04:
Hello,
this is received last time I flashed a KFF device:
~~~
~/devel$ dfu-util -a 0 -s 0x08000000 -D KungFuFlash_v1.09.upd
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 65536
Download        [=========================] 100%        65536 bytes
Download done.
File downloaded successfully
~~~

mamejay, 2021-01-05:
Just had  a thought.  I have had this issue with the resistor ladders not being soldered in correct.  Once run over with an iron and some flux and no more black screen.  Just a thought

C-64Bram, 2021-01-05:
Hi.
Thanks!
I am going to retry building it in a few days and also reworking the
non-working ones.

Best regards,
Bram.

medzes, 2021-02-01:
Just my 2 cents: if you get the error "No DFU capable USB device available" from dfu-utils on Windows, you probably need to install the USB driver for the ST DFU mode. The regular ST tools driver conflicts (is closed source) with what dfu-utils need. You'll need to follow the steps for example here. Short version: 
- Plug in the STM32 usb in your pc, making sure its in DFU mode
- Install Zadig

https://www.hanselman.com/blog/how-to-fix-dfuutil-stm-winusb-zadig-bootloaders-and-other-firmware-flashing-issues-on-windows

faxm0dem, 2021-11-10:
I just had the same issue. I checked all connections on the 2 resistor networks and there was one loose solder joint. Thanks to this thread I found it right away 👍

SunnyIMG, 2023-06-25:
Hello brother, I want to know how much thickness should be selected when making pcb

jgilcas, 2023-06-25:
> Hello brother, I want to know how much thickness should be selected when making pcb

I'm using 1.6mm enig with golden fingers.

Regards

SunnyIMG, 2023-06-26:
> 

thanks,bro,Does the thickness have no effect on reading?

closed 2024-12-29 by KimJorgensen