KimJorgensen/KungFuFlash #36: CR Change Request: change default device ID

corasaniti, 2020-09-26:
it would be very useful to change the device number from 8 to 9,10,11 so as to allow the use of a real drive 1541 set to 9. 
Thanks

mincebert, 2020-10-07:
This would be very useful!  A slightly more niche, but related option for me would be to renumber existing devices: I have a 128D and it would be handy if it could renumber that to something else (say 11) and then use for 8 itself.  I can raise this as a separate issue, if you like.

smdprutser, 2020-11-11:
Since I have a static configuratiion (SX64 with SD2IEC) and no real ASM knowlegde I made a compile time option to change the devicenumber. For personal use I prefer 7 just for the looks :) 

See my pull request for more info #51

KimJorgensen, 2020-11-30:
There is now a configuration setting (F5) where it is possible to change the device number.
@mincebert I'll close this issue, but feel free to open a new one if you would like the option to renumber existing devices.

corasaniti, 2020-11-30:
This is great news tonight as I proceed to update the firmware. Thanks