KimJorgensen/KungFuFlash #37: Win10 drivers for KungFuFlash

pedrouran, 2020-10-02:
Hello,

I'm trying to connect KungFuFlash plugged into C64 via USB. 

C64 is running USB server 1.93.1 downloaded from just here.

Win10 detects the device with no drivers addition. I tried to install EF3 Drivers, but Win10 says that the best driver is already installed.

[image]

But trying to connect via CMD command it holds at thos point:

[image]

I have tried to run EasyTransfer too, but I always get this error:

[image]

Any help will be appreciated.

Thank you

KimJorgensen, 2020-10-03:
Hello Pedro,

You don't need to install the EF3 drivers as Windows sees Kung Fu Flash as a COM port. For now only the modified version of EasyFlash 3 USB Utilities will work with Kung Fu Flash, EasyTransfer and other software for EF3 will not.

One thing to check is what COM port Windows has assigned to Kung Fu Flash. You can see that by right clicking the Kung Fu Flash device, choosing Properties and then selecting the Hardware tab:

[image]

This is the COM port number to use with the ef3usb.exe program.

I hope this will help.
Kim

pedrouran, 2020-10-03:
Thank you for reply.

In my configuration is COM4. The only command i got to work is "ef3usb com4 0" , it makes some "TEST BYTE" in C64 and finishes showing "FF TEST BYTE", and in the console shows "Testing was Succesfull ! Restart C64..."

But I can't get work the rest of the commands like SEND prg or BURN crt, it holds, and I have to type Ctrl + C to abort.

Should those commands work?

[image]

KimJorgensen, 2020-10-03:
You should try to get the "ef3usb com4 s" command to work first. The test command "0" should work after sending the ef3usb.prg file to C64 file via the "s" command.

Is the ef3usb.prg file in the same directory as the ef3usb.exe? Is the C64 in the menu when executing "ef3usb com4 s"? Does anything happen on the C64 screen when executing that command?

pedrouran, 2020-10-03:
Ok, I got it to work:

e[xecute] command
c[opy] command
f[ormat] command
w[rite] command
0 [test] command

Curiosily I could't get work: s command and b command, the firsts commands I was beginnibg to test.

KimJorgensen, 2020-10-03:
If you got these commands to work you must be doing something else on the C64 side than just entering the menu / launcher.
Did you load the "ef3usb.prg" program on the C64 side using the menu?

pedrouran, 2020-10-03:
Yes I load ef3usb.prg 1.93.1-KFF from the menu

KimJorgensen, 2020-10-03:
The "s" command only works if you are in the menu without loading anything. The command transfers the ef3usb.prg from the PC to the C64 and the other commands can then be used.

pedrouran, 2020-10-03:
Ah, ok I see... 

Thank you very much for your help.

I have set Far Manager as well to execute PRG files, and it works very fine. Following this video tutorial https://www.youtube.com/watch?v=2IuOAAEOOeM

KimJorgensen, 2020-10-03:
Good that you got it working

closed 2020-10-03 by KimJorgensen