KimJorgensen/KungFuFlash #38: STM32H7 upgrade idea

Jaystonian, 2020-10-14:
Trying to port this code over from STM32F4 into STM32H7 and ST really made it difficult by renaming or completely changing registers and definitions.  Can I send you an H7 dev board to help?  I'm stuck on the SD interface, working my way through.  The bit of code surrounding "SDIO_CLKCR_CLKEN" for init() and deinit(), this doesn't even exist in H7 and I'm unsure how to handle setting the clock for data write holds.  Also am using STM32CUBEIDE.  Will you help?  The dev board is a big upgrade over the old KFF pcb.

KimJorgensen, 2020-10-15:
No doubt that the STM32H7 is a more powerful microcontroller but it will also drive the cost of the cartridge up. May I ask what you are trying to achieve by porting the code to a STM32H7?

Jaystonian, 2020-10-15:
The cost of the cartridge is still pretty inexpensive compared to everything else on the market.  The STM32H753VIT6 is about $12.

1) This H7 has a max speed of 480MHz, much faster than the F4 used in KFF (you had mentioned it was near its limits for the c64)
2) The extra speed will allow for 2MHz mode for C128 and extra code by the user for manipulating the extra GPIO
3) There is a 256MB NAND Flash onboard the new card as well as 22 extra GPIO for a network expansion or whatever the users want to use it for (perhaps even wiring it back into the c64 for eprom selection or whatever).  The extra memory would allow for 16MB REU (or larger?) and other things.
4) The new card uses *either* the 8MHz standard clock (when disconnected from c64) OR the computer's DOTCLK.  This would allow for the STM32 to be in a phased-lock loop with the computer so the card can work with NTSC machines and improve compatibility with PAL machines.

KimJorgensen, 2020-10-15:
OK. That sounds like a fun project and thanks for the offer, but I'm not interested.
You are welcome to keep this issue open and maybe somebody else following this project would have an interest.

Jaystonian, 2020-10-15:
I've got the project compiling now under STM32CUBEIDE but the SD interface for example has changed since F4 so a lot of the porting was guesswork.  I'm on a 2nd pass now to verify gpio changes etc.  How did you test the card before you were sure it was safe to connect to the 64?

KimJorgensen, 2020-10-15:
I had a development board connected to a C64 and started with a very simple firmware that was only emulating a register that could be written to using POKE in BASIC. An logic analyzer was connected to see if everything was looking correct before attempting writing to the C64 bus.
After read and write was working I then moved on and "ported" the firmware to the PCB I had developed.

KimJorgensen, 2024-12-29:
See also https://github.com/KimJorgensen/KungFuFlash2

closed 2024-12-29 by KimJorgensen