KimJorgensen/KungFuFlash #39: Does not work with 1gb uSD cards

ogg1e, 2020-10-15:
I tried to use a 1GB SD card and it was not recognized.  Is there a size/type of sd cards it does not support?

KimJorgensen, 2020-10-15:
I'm not aware of any unsupported SD card types. Just make sure that you got a FAT formatted partition on the SD card and that you can read/write files to the card using a computer.

ogg1e, 2020-10-15:
I assumed it should work with all cards. But I tried it in two different carts, and it behaves the same. I also formatted it twice to be sure.  

:shrug:

KimJorgensen, 2020-10-15:
Did you try with other cards and does that work? If not it could be a problem with the SD card connector.

ogg1e, 2020-10-15:
I sure did. A 64gb card worked fine.

KimJorgensen, 2020-10-16:
Weird. Did you have a look at the partition table to ensure that there is only one primary FAT partition? Also, what sector size is used?

ogg1e, 2020-10-18:
Yes. There's only 1 partition. I tried formatting it as FAT and FAT32. Neither works. With FAT, it was 16k sector size.  

Like I said, I formatted a 64GB card the exact same way and it works fine.   And this 1GB card works in other devices.

KimJorgensen, 2020-10-18:
16k sector size? FatFs that is used by KFF has these limits http://elm-chan.org/fsw/ff/doc/appnote.html#limits

ogg1e, 2020-10-18:
Sorry.  
`sudo fdisk -l /dev/sdi
Disk /dev/sdi: 982.51 MiB, 1030225920 bytes, 2012160 sectors
Disk model: STORAGE DEVICE  
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x000d8350

Device     Boot Start     End Sectors  Size Id Type
/dev/sdi1        2048 2011135 2009088  981M  e W95 FAT16 (LBA)
`

KimJorgensen, 2020-10-18:
Are these 1GB SD cards available anywhere? I think I need one to be able to find the issue

ogg1e, 2020-10-18:
I doubt it. It's an old one I've had kicking around for a while.   But it's just like this one:
https://www.ebay.ca/itm/Kingston-1gb-MicroSD-card-with-Micro-SD-adapter/143732553233?hash=item217720b611:g:BPYAAOSwqK9fW~hX

KimJorgensen, 2020-10-18:
OK. I'll try to find one

ogg1e, 2020-10-20:
If you can't find one, I could send mine to you.  But then you'd probably be fixing it for nobody. :-)

KimJorgensen, 2020-11-07:
These old cards are surprisingly expensive on ebay, but I managed to find a cheap Kingston 2GB card that I hoped would also fail. I formatted it with the same partition size and type as yours, but it worked fine. So I guess this is only a problem with the 1GB version.

ogg1e, 2020-11-14:
I could mail you the one I have, but the cost might be high as well. I'm in Canada, and I suspect you're in Europe somewhere.

KimJorgensen, 2020-11-15:
Yes, the shipping costs would be more than the price of a brand new SD card. Unless you really want this fixed now I'll park this issue until I can get hand on a card that fails.

ogg1e, 2020-11-15:
Sure. Might as well park it. No sense wasting any more time on this.

closed 2024-12-29 by KimJorgensen