KimJorgensen/KungFuFlash #42: Support for GMod2 cartridges?

modeler, 2020-11-03:
Hello and thank you for this amazing device. At some point in the future, might it be possible for Kung Fu Flash to support GMod2 and similar cartridges? This is similar to the Magic Desk format, but with on-board flash memory for saving game data back to the cartridge:

http://wiki.icomp.de/wiki/GMod2

KimJorgensen, 2020-11-03:
The serial EEPROM would require some effort to emulate but I guess it would be possible

modeler, 2021-01-03:
Just for the record, another popular new game Soul Force by Sarah Jane Avory was released in GMod2 format this week. It works fine with Kung Fu Flash if you change the .CRT header to type "md" (Magic Desk) using cartconv from VICE, but obviously you can't use the save data.

The game is also supplied as a 512K Ocean cartridge, but this does not work with Kung Fu Flash. I tried this image with an stand alone EPROM-based cartridge PCB wired for dumps of Ocean cartridges, and that didn't work either. Wiring it for a Magic Desk cartridge did get it working though.

jaluth, 2021-10-05:
+1 for Gmod2 support. Would love to be able to play the cartridge versions of Hunter's Moon Remastered and Sam's Journy with Kung Fu Flash.

modeler, 2021-10-05:
The 1541-Ultimate and Ultimate64 very recently added support for GMod2, it's been absent for a long time. It would be great to have it supported by Kung Fu Flash, but it's a big ask.

jaluth, 2021-10-06:
> The 1541-Ultimate and Ultimate64 very recently added support for GMod2

I have a 1541 Ultimate II hooked up to my C64, but sadly gmod2 is only supported in the Ultimate II+ version of the recent firmware.

jaluth, 2022-01-05:
On a related note, work is underway on a "GMod3" cartridge format: http://wiki.icomp.de/wiki/GMod3

Would be nice to have that supported by the KFF as well - but as it's not being used for commercial releases yet, I'd be more than happy to "settle" for GMod2 support. :)

oemsysadm, 2022-08-16:
A thousand times yes to this. With save feature.