KimJorgensen/KungFuFlash #44: STM32 Program

Icelvlan88, 2020-11-07:
Hello,

I'm trying to program the STM32 and I keep getting an error. "Connect Under Reset" mode Target -> Settings menu and try again.

I tried programming another STM32 and was able to, so I know my usb is working correctly.

Can you tell me which pins should be connected directly to Boot0, 3.3V, GND, SWDIO, SWCLK and ill try to connect directly to the processor?

I have pin 1 on the STM32 facing C13, so I know the orientation is correct.

I also tried to reflow the board with no luck.

Also, if D1 is not on the board will it still function to program?

KimJorgensen, 2020-11-07:
Hi

Are you using a ST-Link v2? If you got the microcontroller from China there is a chance that the read out protection is enabled. In that case you need a ST-Link v2 and the ST-Link Utility from ST to unlock it.

The schematic for Kung Fu Flash can be found [here](https://github.com/KimJorgensen/KungFuFlash/blob/master/hardware/KungFuFlash.pdf) but it shouldn't make any difference if you connect the pins directly to the microcontroller. D1 is not needed for programming the board.

Icelvlan88, 2020-11-07:
Hello,

I got a ST-Link v2 and the microcontroller is not from China. What is the process to unlock incase? I checked all my continuity and everything looks fine. It says Detection fail on the STM32 settings in the ST-Link utility.

Can you confirm pin 1 on the STM is the corner where C13 is located?

KimJorgensen, 2020-11-07:
The ST-Link Utility will detect if the read out protection is enabled when you try to program the microcontroller and give you the option to unlock.
The microcontroller needs 3.3V which it should get from the ST-Link v2 and the BOOT0 pin should be connected to 3.3V. I guess you have done that, so I would recommend checking the solder joints on the microcontroller with a microscope if you haven't already.

Yes, pin 1 is in the corner where C13 is located. You can see the orientation of the microcontroller [here](https://github.com/KimJorgensen/KungFuFlash/blob/master/pics/pcb_rev1.jpg)

Icelvlan88, 2020-11-07:
For the programming portion, does any other pin need and or 3.3v?

KimJorgensen, 2020-11-07:
You only need to connect 3.3V to to the J4 pins labeled "+3.3V" and "BOOT0".

Icelvlan88, 2020-11-12:
I soldered 3.3V and BOOT0 together. Might have a bad chip. Ordered a new one.

Icelvlan88, 2020-12-13:
Was able to program, but I have an NTSC C64. Any idea when you'll be able to add support? Is it a software or hardware thing? When the cart is inserted, the system won't turn on.

KimJorgensen, 2020-12-14:
NTSC support is a software issue that is being worked on. There is already an issue for this (see #6) so I'll close this one.

closed 2020-12-14 by KimJorgensen