KimJorgensen/KungFuFlash #47: API via command channel for direct access to SD Card

lubber-de, 2020-11-08:
## Feature Request / Idea
Do you think it will be possible to implement some little API via the command channel 15 to allow access to the SD card (to load/write single files directly on SD  / switch disk, etc) ? 
As you already have this available somehow via the menu button, i was hoping for a possibility to expose some of the used methods to an API to be accessible to c64 programs directly.

I am thinking about a similiar functionality like UltiDOS for 1541u-ii / u64

KimJorgensen, 2020-11-08:
Yes, I think that would be possible.

dikdom, 2020-11-13:
Maybe using sd2iec-like (same?) commands..

medzes, 2021-05-01:
Yes, this would be possible, but requires some refactoring of existing code then adding a set of calls like the sd2iec code has.
The main issue I see is that the added ARM code size will likely not fit in the current available space.

KimJorgensen, 2024-12-29:
There has been support for basic SD2IEC commands since [firmware v1.33](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.33)

closed 2024-12-29 by KimJorgensen