KimJorgensen/KungFuFlash #48: Does it work with a REU via Port expander?

lubber-de, 2020-11-08:
## Description
I wonder if the KFF will work together with a real REU via Port expander?

I understand that it wouldn't, when an EasyFlash Image is emulated (because EF has extra RAM at $df00-$dfff which is also occupied from a REU).
But when other cartridge types are emulated, which do not cover $df00: Would it work with a real REU via Port expander?

KimJorgensen, 2020-11-08:
The Kung Fu Flash was not developed with a port expander in mind and this is not something that I have tested. So I don't know.
I would not recommend trying this but if you do, please note that the menu runs in EasyFlash mode.

lubber-de, 2020-11-08:
Ok, then i guess, it won't work if the menu already runs in EF mode. Thanks for the answer.

closed 2020-11-08 by lubber-de

mincebert, 2020-12-20:
I have my Kung Fu Flash in an X-Pander 3 and it appears to work fine with a 1750 REU on my 128DCR.  I haven’t done many tests, though, but the REU test disc ran OK and I managed to use TurboMacroPro with REU + Kung Fu Flash as an Action Replay and also Super Snapshot.  The Kung Fu Flash menu seemed to work OK, too.

KimJorgensen, 2020-12-20:
Thank you for the feedback

drphilxr, 2023-02-14:
Got my X-pander 3 - and have the 1764 REU, and the KFFlash.
Put the 1764 into middle slot - using I/O 2- with KFF in the slot closest to cart port (#1?)
and can run the reu ram test globe - but not sure at all of how to get KFF hooked into rom.
The I/O 1/2 jumpers confuse me at each slot ("swap") and the manual assumes you're Jim Butterfield and have a C64 mem
map in your mind...