KimJorgensen/KungFuFlash #52: KFF not sending EF3 prg.

mwedmark, 2020-11-15:
I just managed to get my KFF to work with the C64. The only thing left is the micro-SD connector which I ordered the wrong variant of. That will be fixed soon. But after flashing 1.14 I already see the "Please Inser a FAT formatted SD Card" prompt on my breadbox, so that's nice!

I though I would start by trying out that KFF version of EF3 remote execute from PC. I can see that the KFF got COM3 when connected.

I try "EF3USB.EXE COM3 s" which shows the following (see below) but nothing happens. Do I have to have a working micro-SD card connected to get the EF3 receive to work or am I missing something else? 

```
EF3 USB Client v1.93.1 - Kung Fu Flash version

 - SEND PRG TO EF3 MENU

Handshake: EFSTART:PRG
```

KimJorgensen, 2020-11-15:
Yes, a working micro-SD card is need to get this to work

mwedmark, 2020-11-15:
ah ok thanks for the quick answer! Good to know! But micro-SD is not used if just do execute a file on C64 f.e. (ef3usb.exe COM3 e file.prg)? Maybe I misunderstod, I though I was going to be uploaded directly to C64 RAM via the bootcode in STM32.
I will fix that promptly!

KimJorgensen, 2020-11-16:
Without a working SD card the cartridge will not start and cannot accept commands from the USB port and send these to the C64.

closed 2020-11-17 by mwedmark