KimJorgensen/KungFuFlash #53: The Three Stooges (EasyFlash release) not working.

modeler, 2020-11-17:
Of all the EasyFlash releases I've tried with Kung Fu Flash, this is the only one that doesn't work (which is pretty amazing overall):

https://csdb.dk/release/?id=189629

Just goes to a black screen after the intro, when it should jump to the "Defender of the Crown" joke intro. The image works in VICE and with Ultimate64, I don't have an EasyFlash cartridge to try it with though. This is a very minor thing, just thought it might yield some info on a potential compatibility issue.

Thanks!

KimJorgensen, 2020-11-18:
I cannot reproduce this issue.
The "Defender of the Crown" screen is shown after ~8 seconds which is the same behavior as with my EasyFlash 3 cartridge.

modeler, 2020-11-19:
Thank you for having a look Kim. This is really strange, I have tried the image on all four Kung Fu Flash cartridges I have built so far and haven't been able to get it working. I have tried building the latest firmware from Github also.

I have many different C64s at my disposal right now as I repair them, pretty much one of each board revision. They all behave the same for me, just a black screen after the 'cracktro'. I have no other peripherals attached, just the Kung Fu Flash and a joystick.

hackup, 2020-11-19:
That is a little strange. I tried that CRT on a random C64 that I had on the desk (using a KFF, of course) and it worked just fine for me.

modeler, 2020-11-19:
The Cinemaware Collection that was released alongside it (even has the same intro) works fine with my KFF cartridge: 

https://csdb.dk/release/?id=103750

But The Three Stooges still doesn't work at all. Of the 20 or so EasyFlash releases I have tried with my self-built KFF, literally all of them work perfectly apart from this one. I don't see what I could have done differently in the build that would explain this, so I would be very grateful if anyone else can test this image for the sake of my sanity.

Thanks!

modeler, 2020-11-19:
Good news, I was able to get the image to work but _only_ if there is a disk drive connected. This is of course an issue with the crack used to create the EasyFlash release and nothing to do with the Kung Fu Flash itself. Apologies, I thought I had covered this off before raising this issue. The mistake I made was plugging in my Pi1541 without mounting a blank disk i.e. I hadn't put it into emulation mode. Similarly, connecting an original 1541 drive but leaving it switched off isn't enough, the program must execute some drive code as it loads as the drive has to be on.

I can only assume you guys both have a drive connected. I was able to reproduce the behaviour in VICE, the image fails to start in the same way if I simply disable drive 8. Closing this, thanks again for all your efforts!

closed 2020-11-19 by modeler

hackup, 2020-11-19:
Yes, I did indeed have an SD2IEC connected at the time.

KimJorgensen, 2020-11-19:
Yes, I also had a SD2IEC connected. Good that you found the issue