KimJorgensen/KungFuFlash #54: KFF Firmware Build description/guide

mwedmark, 2020-11-18:
Sorry to create entries here (but at least I remove them after answer..)
Is there a description how to install and run the simplest toolchain to build KFF? I tried installing System Workbench for STM32 in windows and tried importing the project. Did not work.. Also tried several ways of setting up the tool-chain within that UI but failed. Also read that VSCode can be used. Maybe someone has a VM in which a build is possible, maybe a Docker container for build or a simple description to get it to work. Thanks! Mostly for fun but of course I will produce pull requests if any good comes out of it.

KimJorgensen, 2020-11-18:
There is no description I'm aware of. I'm using Linux for development but I guess that it would be possible using Windows as well.
Tools: gcc version 9.3.1 (GNU Arm Embedded Toolchain 9-2020-q2-update) and cc65 V2.18

hackup, 2020-11-18:
I can confirm that the project will build on Windows as well, using the same tools.

mwedmark, 2020-11-20:
OK thanks guys! I managed to get it working. installed cc65 via choco installer and System Workbench STM32. After adding some paths everything worked a charm! Now I can build and extend KFF firmware,

closed 2020-11-20 by mwedmark