KimJorgensen/KungFuFlash #56: Expected result with Ultimate64?

modeler, 2020-11-19:
While trying to make sense of https://github.com/KimJorgensen/KungFuFlash/issues/53 I tried Kung Fu Flash in my Ultimate64, which didn't boot. It just hangs at the initial black screen. No damage done to either hopefully.

Although it makes no sense to use a cartridge like KFF with U64 on a day-to-day basis, there are scenarios such as this where it would be useful to load from it. I would be interested to know if anyone else has tried it, whether or not it is expected to work or if there's a logical reason why it wouldn't.

Thanks!

dillof, 2020-11-19:
Which firmware is your Ultimate 64 using? The latest (1.34) is known to be less compatible with cartridges than the previous one (1.28). E. g. IDE64 works with 1.28 and just hangs with a black screen with 1.34.

modeler, 2020-11-19:
It's the latest V1.34, thanks for the info.

modeler, 2020-11-30:
Closing this, the consensus being that U64's cartridge compatibility is hit and miss in general.

closed 2020-11-30 by modeler

modeler, 2021-04-28:
For the record, today I found that Kung Fu Flash appears to work fine with Ultimate64 from firmware version 1.17 onwards. It is not mentioned in the release notes, but might be worth remembering.