KimJorgensen/KungFuFlash #59: txt file reader?

Sebby75, 2020-11-27:
Any chance to implement txt file reader into launcher/browser??

With most of my collection crts (eg, game collection, utility collection etc) I have small description txt files listing what's on it, it would be handy to be able o read it before loading/flashing actual cart... I'm finding myself loading collection crt only to realise that what i'm looking for isn't there...

which brings me to another question.. what is expected write cycle count of flash memory of stm32f4 ? 1000? 5000? 10000? writes.

Regards
Seb

KimJorgensen, 2020-11-27:
https://community.st.com/s/question/0D50X00009cga5RSAQ/stm32f405vgt7-flash-memory-write-endurance-how-many-times-can-the-internal-flash-be-written-erased

KimJorgensen, 2023-07-27:
A text file reader has been added in the [1.46 release](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.46)

closed 2023-07-27 by KimJorgensen

corasaniti, 2023-08-05:
@KimJorgensen 
I created a file with the .txt extension but from the browser I don't see it inside the folder? How can I resolve?
Thanks

Sebby75, 2023-08-05:
[image]

Seem to be working perfectly for me... created .txt file in notepad on pc moment ago and tested...
Are you sure that file is actually there??