KimJorgensen/KungFuFlash #63: Override cartridge boot.

palehub, 2020-12-08:
Would be possible to implement a special key (RUN/STOP , CTRL ...) that when pressed while powering/resetting the C64 would override the KFF from starting? Or at least make it go direcly to the menu? Sometimes I do not like my C64 powering on directly to the last used file.

Would be possible to define certain keys in such way to start some user-predefined .crt files? For example, booting to FinalCartridge when booting while pressing 1, and to SuperSnapshot while pressing 2, and so on?

KimJorgensen, 2020-12-09:
If you would like to disable KFF you can press F8 in the menu and if "Persist BASIC selection" is enabled in the settings (F5) then the KFF will be disabled until you press the menu button and make a selection.
I'm not that keen on prolonging the startup sequence to wait for a key press or having to execute code on the C64 before starting the cartridge which reduces compatibility.

I will label user-predefined .crt files as an enhancement

CqaZ80, 2022-09-27:
maybe a "persist kff Menu selection" option could be done.