KimJorgensen/KungFuFlash #64: Any help running "Ghouls'n'Ghosts (US Gold)"

G7200, 2020-12-16:
I cannot run the following game "Ghouls'n'Ghosts (US Gold)". I use the latest firmware.

KimJorgensen, 2020-12-17:
Is it a D64 or a EasyFlash release? Could you please provide a link to the release?

G7200, 2020-12-17:
Hi, thanks for your reply.

I meant just the 'normal' D64 diskfile. There are many releases of this
game but sadly none of them work with the KungFuFlash. All I get is a black
screen.
Normally it takes a while to load up that fantastic soundtrack in the main
screen but unfortunately that isn't happening.

Here's a few ones I've found online to test:
http://www.commodoreserver.com/PublicDiskDetails.asp?DID=249C1FDAF4D0439483C3448C2427D14E

or

https://csdb.dk/search/?seinsel=all&search=ghouls%27n&Go.x=0&Go.y=0

I can load it on my old sd2iec but it would be awesome to get it loading on
the KungFuFlash too which is of course much quicker.
Greetings & thanks!
;) Balduin

KimJorgensen, 2020-12-18:
D64 support will never be perfect, but you can try this [EasyFlash conversion](https://www.forum64.de/index.php?thread/47989-disk2easyflash-fertige-images/&postID=1546500#post1546500) by Metallic. Requires access to forum64, though

G7200, 2020-12-18:
oh great, i will try that out, thanks!
:) balduin

G7200, 2020-12-18:
Hi Kim,
And it works, many thanks for your help!
I love this port and game soundtrack
all the best from switzerland,
;) balduin

KimJorgensen, 2020-12-18:
Great then I'll close this issue :)

closed 2020-12-18 by KimJorgensen