KimJorgensen/KungFuFlash #66: RESET and MENU Issues

Pr0spect0r, 2020-12-20:
Hello Kim

I'm enjoying your great project, thank you. 

Need your assistance, I've ordered the PCB from JLCPCB. I'm having an issue with the board where it hangs the C64, when the RESET, MENU or the RUN/STOP is pressed. Are you aware of anyone else having these issues? Is there a solution you might be aware to resolve the RESET issue?

FYI: I'm using a Breadbin C64, and I can read and run any C64 game or program without any issues, it's only when I want to reset the C64 to load another game it hangs.

Regards,
Naji

KimJorgensen, 2020-12-20:
Hi Naji,
No, I'm not aware of anybody else having these issues.
If reset doesn't work correctly then I would check that the correct components are used for R3, R4 and Q1.

Pr0spect0r, 2020-12-21:
Hello Kim,
Thanks for the heads-up... it was the Q1 (BC817) not functioning correct, replaced it - and all is well now. 

Keep up the great work.

closed 2020-12-21 by Pr0spect0r