KimJorgensen/KungFuFlash #67: Which version of SIDPlay64 work with KFF?

mwedmark, 2020-12-20:
I though I'd try a detour with my KFF SID Player and build my own local SIDPlay64. Maybe that would be a possible solution also. But mostly for fun. As the end result I would like to either build a EasyFlash compatible SIDPlay64 or a specific version for KFF which would read SIDs from SD-card.

A first try is to try make one of the standard ones work within a D64 files, which version whould I use? No version seem to work out-of-the-box.

The versions available seem to be:
Normal - Standard 1541 access
1541u2cmd - Ultimate 2 support?
iec-cmd-ide64 - HD support?
SD2IEC - SD2IEC support
1541u-netdrive - Ethernet Mounted network drive??
chameleon-normal - ??
chameleon-sd2iec - ??

On-top of that support for: SIDFX, REU and them some..

I'm currently trying to build the "normal" version. While building it also creates a test D64-image which try to find files at device @8. I can see those files in the browser within KDD menu, and can also start the sidplay.prg inside the D64 image but it does not find any files. Any idea why? Maybe I misunderstood the D64 support within KFF.

KimJorgensen, 2020-12-21:
SIDPLAY64 seem to use the command channel and load the directory using the open vec

mwedmark, 2020-12-21:
Ahh thanks.  I don't know that kind of detail around the 1541. So all the current versions are incompatible with KFF as of today. Thanks for taking the time.

closed 2020-12-21 by mwedmark

Seni1ix, 2021-12-27:
Any update on this? been trying to find a way to play SID files, and native support would have been awesome :)

mwedmark, 2021-12-27:
Sorry to say no. I would also like native SID support. I'v started a local (naive) version of SID player which actually "works" playing static SID already included in image. Bu when trying to load it from disk, something bad happens. I didn't get to a point where I could actually get it to work. It includes its own play routine (found on internet) and is currently not relocatable. If there is any interest in continue that pursuit I could try and fork code and give link,

KimJorgensen, 2022-01-05:
sidplay64-sd2iec will work with the [latest firmware](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.33)