KimJorgensen/KungFuFlash #68: REU emulation

matti157, 2020-12-22:
Will it be possible to emulate a REU in the future?

KimJorgensen, 2020-12-22:
I think that is unlikely with the current hardware. The microcontroller only has 192kB of RAM and a large portion of that is used by the firmware to speed up execution. Another issue is that the DMA line is not connected and there are no free pins on the microcontroller.

matti157, 2020-12-23:
Forgive me for writing nonsense, but I'm not a very experienced programmer :)

1-If the CPU has even only 16K free it can be interesting to have the REU functionality even just to give some more features to the Kung Fu Flash

2-By creating a special empty file on the SD, can't that be used as virtual RAM? Too slow compared to the 80s RAM that Commodore used?

Anyway I bought a KFF, congratulations it's fantastic!

smdprutser, 2021-01-02:
how many is exactly free? From a quick glimpse of the source code I would say 64+32K?

KimJorgensen, 2021-01-03:
Yes, that is right

beamrider66, 2021-12-19:
I expect with the release of Sonic 64 people will be looking for something like KFF but with REU support.

KimJorgensen, 2024-12-29:
See [Kung Fu Flash 2](https://github.com/KimJorgensen/KungFuFlash2) for REU support

closed 2024-12-29 by KimJorgensen