KimJorgensen/KungFuFlash #69: EasyFlash game compilations crash on my 205469 SixtyClone build

tyristori, 2020-12-25:
Only the compilations that use EasyFlash menu crash. Everything else that I've tried seems to work fine. Here's the video of the issue:

https://user-images.githubusercontent.com/68354113/103137736-38823300-46d4-11eb-8376-8727c55d2d15.mp4

I'm not saying it's a bug of Kung Fu Flash. I have original 205469 C64C and these compilations don't crash on it.

If anyone has build 205469 SixtyClone  I would appreciate if you could try this crt on your machine to see if it crashes:

https://drive.google.com/file/d/1-PElGKR3dfRGJxNc540KC_SWEz0UEdKj/view

KimJorgensen, 2020-12-25:
Thank you for reporting. Did you try test that compilation with an earlier firmware version e.g. 1.16 or 1.17, and what was the result?

tyristori, 2020-12-25:
Yes I've tried 1.16 and 1.17 and the same thing happens.

KimJorgensen, 2020-12-26:
OK, thank you for testing. It is good to know it is not related to the latest changes I made to the firmware.
Another thing to try is to start the CRT in VIC-II/C128 mode. You can select that in the options menu (shift+enter or long press fire).

Are you using a stock kernal and PLA in the 205469 SixtyClone?

tyristori, 2020-12-26:
Starting CRT in VIC-II/C128 mode didn't make a difference.

I'm using original MOS/CSG kernal and PLA.

tyristori, 2020-12-28:
I found the problem and as I suspected it had nothing to do with Kung Fu Flash. I had soldered RP3 resistor network the wrong way and that caused the crash.

Sorry for waisting your time.

closed 2020-12-28 by tyristori