KimJorgensen/KungFuFlash #70: Using DfuSe

userc64, 2020-12-31:
I am trying to program using dfuse. Are there any instructions on how to do this?

KimJorgensen, 2021-01-03:
Have you seen this? https://github.com/KimJorgensen/KungFuFlash/blob/master/firmware/README.md#usb

userc64, 2021-01-03:
Yes I did. That is where I found out you can use Dfuse to program. BUT the issue I am running into is how to convert the .upd file into a .dfu file that is used by Dfuse. I have tried simply changing the file extension and also converting with the dfu file manager. I don't see any instructions on the page you link to that addresses this. Maybe I am not seeing it?

KimJorgensen, 2021-01-03:
Sorry, there are no additional instructions available as I don't recommend using this tool.

You will need to use the Dfu file manager to convert the .upd file to a .dfu file.
- Click "Multi BIN..." and select a .upd file by changing "*.bin" to "*.upd" in the dialog
- Enter the address 08000000 and add to list
- Then generate the .upd file. You don't need to change device settings.

You can now use the DfuSe Demo application to program the KFF

userc64, 2021-01-03:
Worked perfectly! Thank you. I was adding in an additional step that wasn't needed.  I was running the file through the UPLOAD ACTION also before using the UPGRADE OR VERIFY ACTION.

So the steps are quite simple.
Open the Dfu file manager
Choose the top option GENERATE A DFU
Select Multi Bin
Change SAVE TYPE AS to All files (*.*)
Double Click the .upd file
Set address to 08000000
Add to list
Generate
Plug in KKF via USB in DFU Mode
Now open Dfuse
Bottom right Click CHOOSE
Double click the .dfu file you just created
Click Upgrade.

That's it.

Thanks!

closed 2021-01-03 by userc64