KimJorgensen/KungFuFlash #71: Anyone managed to get Soul Force CRT working?

mamejay, 2021-01-04:
Anyone managed to get Soul Force CRT image working?
After the title screen I get garbage on the screen.  
Thanks

modeler, 2021-01-04:
Yes, there are a few ways. Are you using the SoulForce_ocean.crt image? If so, this has the same issue when using a stand alone cartridge PCB. It works if the cartridge is wired for a Magic Desk image, suggesting that specification for this image is off. Don't use it.

The official download from ProtoVision was updated with a new image named SoulForce_GS64.crt, which is based on the C64GS cartridge. This works fine with Kung Fu Flash.

The GMod2 image isn't usable as Kung Fu Flash doesn't support this format, see https://github.com/KimJorgensen/KungFuFlash/issues/42.

The Soul_Force_EasyFlash.crt image does work, but there is another bug where it hangs when writing data back to the cartridge, which I am raising separately.

kaloun34, 2021-01-15:
Hello,
The c64gs version of the game works

closed 2021-03-14 by KimJorgensen