KimJorgensen/KungFuFlash #72: Soul Force (EasyFlash) hangs intermittently when saving or erasing game data.

modeler, 2021-01-04:
The EasyFlash version of Soul Fource (a [new release](https://protovision.itch.io/soul-force)) appears to hang when writing back to the cartridge. This can be triggered one of two ways:

1. From the game's menu, navigate to Options > Data Menu > Factory Reset > Reset All Game Data? > Yes.

[image]

2. Play through the first level, kill the boss that looks like RoboCop's torso.

[image]

An EasyFlash 3 user has confirmed that the issue does not occur on the original EasyFlash hardware.

Kim, if you don't have this game already, please let me know and I will gift it to you.

KimJorgensen, 2021-01-05:
Thanks, I have purchased that game yesterday to debug on this issue.

nikosidis, 2021-01-07:
For me the game can hang after finishing a level and most of the time after writing high-score it hangs. Like reported it seams that writing data is the problem.

KimJorgensen, 2021-01-07:
Yes, I have reproduced the problem where the game hangs after a Factory Reset using both a Kung Fu Flash and an EasyFlasy 3. To me it seems that the problem is with the game, not the save implementation in Kung Fu Flash

markusC64, 2021-01-09:
I also have looked at the game crashing in order to improve Ultimate 64 / 1541 Ultimate II(+).

I found that timing is most likely the cause for crashing. I. e. the time needed to write one byte ($df80) and the time to erase a sector ($df83).

In order to get a better understanding of Soul force hanging: Do you know how long KFF takes for these functions?
For the U64/U2(+), erasing a sector e. g. has not constant time, but a probability distribution of the time needed.

KimJorgensen, 2021-01-09:
Hi Markus,

Knight Rider and I have been working together to get to the bottom of this.

I too noticed that the crash seemed to be timing related when I started to add debug information to KFF's eAPI implementation. So I made a dummy eAPI driver that always return OK back, without writing anything to flash and modified VICE-3.5 to use that driver instead of the default one.
Using that driver we could reproduce the crash in VICE and see that the game was stuck in this look forever:
```
.C:082c  C5 F9       CMP $F9
.C:082e  F0 FC       BEQ $082C
```
According to Knight Rider, the problem was that bit 9 of D012 was set, and the IRQ that INC $F9 never was executed. He has fixed the bug but I'm not sure when a new release of the game will be out.

markusC64, 2021-01-09:
Yes, that's the point. The dummy EAPI also modifies timing... I have seen that loop in Vice to - emulating the 1541 U2 EAPI (my modified VICE can do this with the exception of "erase sector" timing.

Anyway, I'd like to collect timing from different Easyflash Implementations so they can be emulated - and thus enable c64 coders to test with that timing in a patched emulator. If you havn't, and also no idea how to meaure, then I'll have to look for another way to find out.

modeler, 2021-01-09:
I was going to apologise for wasting your time, since I only raised it as a Kung Fu Flash issue because I was told it did not happen on EasyFlash 3 hardware. These efforts to debug the issue for the benefit of other developers if commendable, thank you guys.

KimJorgensen, 2021-01-09:
@markusC64 No, I haven't done any precise measurements and don't have the possibility to set break points. I guess you could develop a C64 program to do the this, but to be honest, I don't think it is a good idea to have uses of the EAPI depend on the timing of the different implementations.

KimJorgensen, 2021-01-09:
@modeler No problem at all. With a bug like this it is not always apparent where the problem lies.

markusC64, 2021-01-09:
@KimJorgensen I agree. But just think the Soul Force develeoper would had the possibility to emulate the 4 APIs Standard Easyflash, 1541 Ultimate, TC64 and KFF. Quite probable, the problem would habve been found before release. That's the idea.

markusC64, 2021-01-09:
But no problem at all - I think the Ultimate 64 will support me dooing my own measurements, when tthe KFF is delivered. You see, it supports sending expansion port data (that is address bus, data bus and some signalling) in real time over ethernet to a pc for diagnostics. Should be possible to use this for time measurements.

nikosidis, 2021-01-10:
From what I understand this game is only tested under emulation?
To me that seams strange. How hard is it to get a betatester with real hardware?
From all the .crt files I tried it is only this one I found problems.

KimJorgensen, 2021-01-10:
@markusC64 If you agree that uses of the EAPI shouldn't depend on the timing of the different implementations, couldn't you just emulate two: A very fast and a very slow implementation? That would also have reviled the problem.

If you want to do measurements of KFF and emulate that, just note that the timing for EF banks 0-3 is different from banks 4-63.

markusC64, 2021-01-10:
@nikosidis No one said that the game has not been tested on real hardware. But it looks like that it had not been tested on 1541 Ultimate II(+), Ultimate 64 and Kung Fu Flash.

markusC64, 2021-01-10:
@KimJorgensen In general you're right. But I have been using the emulation also to find real EAPI incompatibilities. You see, having VICE monitor to debug is very usefull. Still not ready to give up this - and emzulation would be more perfect with right timing.

KimJorgensen, 2021-01-10:
@markusC64 Yes, I agree that having the VICE monitor to debug is very useful and it was the very reason for making the dummy EAPI driver.

markusC64, 2021-01-14:
> Knight Rider and I have been working together to get to the bottom of this.
[...]
> According to Knight Rider, the problem was that bit 9 of D012 was set, and the IRQ that INC $F9 never was executed.

He is absolutely right. I have patched Soul Force's access to that register and it's working well.

PS: You have noticed that bit 9 on a 8 bit system is strange? The answer is that the most significant bit (you may number it is 7 or 8, depending whether you start countijng at zero or one) of $d011 is another bit of $d012 from the meaning of that registers.

nikosidis, 2021-01-16:
Do you know when the game will be updated with the fix?

mrr19121970, 2021-01-21:
Please apply this patch https://mega.nz/file/s0l2USwI#qkmgvBcMRzh1Mm5MmsRRA9l3TyaB5rFr4py8PMvpw6g

nikosidis, 2021-01-21:
Thank you Kung Fu masters :D Works great :D

KimJorgensen, 2021-03-14:
I'll close this as this is not a Kung Fu Flash issue and there is a patch available

closed 2021-03-14 by KimJorgensen