KimJorgensen/KungFuFlash #73: Wormhole D64 problem

nikosidis, 2021-01-07:
This game:
https://www.protovision.games/games/wormhole.php?language=en

If 1541 is connected the games does not load at all.
If I disconnect 1541 it stars to load. I get pass the protovision logo, I see the picture of the game loading. I get some gfx corruption at the lower part of that picture. After some loading the computer hangs.
When I load action replay from KungFuFlash and start the game from 1541 everything is good. 

All the same I'm very happy with KungFuFlash. Great job you done there :)

Sebby75, 2021-01-07:
Game is working for me.
C64C, KFF 1.20pal firmware, external 1571 set as drive 8 and kff as drive 9.
Mounted d64 in kff and game loaded just fine, there is that flickering bottom part of loading screen.. then screen goes blank maybe for 3 seconds and I get into game menu.

nikosidis, 2021-01-07:
Thanks for the info Sebby75.
I have not set KFF as drive 9. Is that default?
Seams there are something there since "Wormhole" does not load with 1541 connected.
It is still not right the flickering, gfx corruption in the loading screen.

Sebby75, 2021-01-08:
KFF disk drive emulation is basic and multi load games that use fastloaders are unlikely to work...

If it works, it works.. if it doesn't, then stick to single load games and games in cartridge format or use KKF to write d64 floppy image to real floppy disk in your 1541 drive and play from there.

nikosidis, 2021-01-08:
In the Wormhole package there is a cartridge directory. Problem is that it is not .crt file. It is 3 files, 1 bat and 2 bin files. I know this can be used from VICE, but is there a way to join them as a single .crt file?

Sebby75, 2021-01-08:
If I'm not mistaken, you can also convert wormhole-gmod2-flash.bin that you have received after purchase to easyflash cart format with cartconv.exe that comes with VICE emulator (it's in bin folder).

https://vice-emu.sourceforge.io/vice_15.html#SEC323

mrr19121970, 2021-01-21:
Not sure cartconv will change a gmod2 to an easyflash

KimJorgensen, 2021-04-11:
I assume the problem with the D64 was due to the limited D64 support so I'll close this issue

closed 2021-04-11 by KimJorgensen