KimJorgensen/KungFuFlash #74: EasyFlash BBS support?

mwedmark, 2021-01-12:
I just scanned the net for fun stuff with EashFlash connection and found this:
[https://csdb.dk/release/?id=159907&show=summary](https://csdb.dk/release/?id=159907&show=summary)

Telnet for C64 using the EasyFlash USB to PC and this specific C64 program. Sound like fun, but dos not seem to work with KFF. Anyone knows what's missing? My guess would be that there may be some difference in the strings sent in the protocol.. I can't find the source for the hack.

Another variant is this:
[https://github.com/sblendorio/petscii-bbs](https://github.com/sblendorio/petscii-bbs)

Anyone got that working?

Of course, the real usage of this is limited so just for fun and more usage for our KFF.

KimJorgensen, 2021-01-14:
Yes, that would be fun.

As written in the [readme](https://github.com/KimJorgensen/KungFuFlash#usb-port) the program on the PC side must be modified to support Kung Fu Flash. If you could get Tom-Cat to release the source code for the PC side, then it could be modified just like I did with EasyFlash 3 USB Utilities.

KimJorgensen, 2021-08-12:
A modified version of EasyFlash3 BBS 1.0 that works with Kung Fu Flash has been added to the repository. See https://github.com/KimJorgensen/KungFuFlash/blob/master/3rd_party/ef3bbs

closed 2021-08-12 by KimJorgensen