KimJorgensen/KungFuFlash #77: New PCB revision for backwards pointing buttons ?

uffejakobsen, 2021-01-22:
First KFF is becoming a really nice project :-)

I've on several occasions experienced that pressing the cartridge buttons tend to "wriggle" the cartridge in a way that the system crashes (freezes).

A qualified guess is that the "wriggle" happens due to the upward pointing location of the buttons.
If the buttons were backwards pointing this would not happen.

This is observed from time to time with two KFF carts in four different systems - and I must admit that it is annoying that the user has to be careful about how he/she presses the buttons and/or need to support the cartridge with standoffs on the backside.

I know that the backwards pointing buttons has another mounting-layout - so my suggestion would be to incorporate this into any planed revision of the PCB.

KimJorgensen, 2021-01-25:
I have experienced the same thing on one of my boards but cleaning the connector did resolve the problem. So I'm wondering how many has this problem and if cleaning the connector would also help them?

uffejakobsen, 2021-01-25:
> I have experienced the same thing on one of my boards but cleaning the connector did resolve the problem. 
> So I'm wondering how many has this problem and if cleaning the connector would also help them?

Well, I have tried that also - but I still see crashes caused by pressing the buttins from time to time...
I guess that our C64's are just getting old (they are) and the expansion port have been in used and worn for over 35 years.
Most other cartridges have backwards pointing buttons - and they do not give me the same trouble when used with the same boards

rickard-von-essen, 2021-01-25:
A device like this is helpful:

https://www.ebay.com/itm/Tombstone-64-Commodore-64-reset-board-and-expansion-port-angle-adapter-combo-/283816694848

uffejakobsen, 2021-01-25:
> A device like this is helpful:
> 
> https://www.ebay.com/itm/Tombstone-64-Commodore-64-reset-board-and-expansion-port-angle-adapter-combo-/283816694848

Thanks - but I already own several C64 expansion port extenders :-)

That will only move the problem - you are still pressing the buttons in the wrong direction

Also using an expansion port extender in order to fix this issue seems a little weird...
And it won't certainly help to the success of this device - if a subset of the users will have to apply such "patches" to KFF in order to get stable operation - that should come from the device itself.

pmandes, 2021-03-20:
Hi,
I cloned the main repo and made a separate branch with a new PCB that has right angle tactile switches:

https://github.com/pmandes/KungFuFlash/tree/long-board-cart

...still a work in progress, but I'm almost at the finish :)

kroesi, 2022-02-12:
> Hi, I cloned the main repo and made a separate branch with a new PCB that has right angle tactile switches:
> 
> https://github.com/pmandes/KungFuFlash/tree/long-board-cart
> 
> ...still a work in progress, but I'm almost at the finish :)

Hello @pmandes,
thank you for creating this fork. Did you build it? Does it work?
I just wanted to order it at jlcpcb with assembly, but the STM32 was to expensive for me at the moment without knowing if it will work.
Krösi

pmandes, 2022-02-12:
Yes I built it, tested it and it works very well, however I didn't make the SMT assembly files so you have to either assemble yourself or generate these files yourself with KiCad.

closed 2025-03-08 by KimJorgensen