KimJorgensen/KungFuFlash #80: Emulate Freeze Frame cartridge

gold11-carca, 2021-01-27:
Good evening the community
Could it be possible for the KFF to emulate the Freeze Frame cartridge a full wiki is available here https://rr.pokefinder.org/wiki/Freeze_Frame

Thank

KimJorgensen, 2021-02-06:
Support for Freeze Frame and Freeze Machine has been include in [version 1.21](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.21).

Please note that some of the old CRT files from pokefinder are in a wrong format and should be recreated with cartconv using the .bin files, otherwise they would not work correctly with KFF. Information about the wrong format can be found [here](https://sourceforge.net/p/vice-emu/code/HEAD/tree/trunk/vice/src/c64/cart/freezemachine.c#l254).

closed 2021-02-06 by KimJorgensen

gold11-carca, 2021-02-07:
Hello,
Thank you very much for responding to my request, the emulation works perfectly.
Thank you for your work KIM