KimJorgensen/KungFuFlash #81: Unstable sdcard reads

medzes, 2021-02-01:
Hi all,

I have a situation that my sd-card is frequently (25% of the time) mis-read and gives read errors at random times, like when loading the next page of directory entries. Anyone have an idea if there is some timing related issue or some STM32-configuration related setting in the code I could try to change to see if that helps?

KimJorgensen, 2021-02-02:
Based on what you have written elsewhere I'm assuming that you not are seeing these issues with the Kung Fu Flash firmware but are using some derived firmware that you developed.

If you have changed the firmware in such way that you are spending more time on interrupts then you could get timeouts or buffer overruns when reading or especially writing data to the SD card. You could try enabling error logging in diskio.c

medzes, 2021-02-24:
I have finally been able to spend time tracking down the reason for the sdcard issues I was having. Fortunately I've been able to solve them and make KFF work rock solid. 

I enable the hardware flow control bit in the SDIO peripheral's control register. After this, I have not had trouble with buffer underruns anymore.
I also implemented support for the 1/4 fifo full bit in the  SDIO->STA status register. When that is high an do 8 fifo reads at once without checking status again.
I also added some __DMB()'s after SDIO register writes, to help lower interrupt lacency.

My main guess why I'm having more issues, where all I changed in KFF is the GPIO ports used, is that the compiler version just happens to have small difference that can have nasty side effects on interrupt latency jitter. Minute code changes to the interrupt routine can cause the C optimizer to take a slightly different path, with a (comparitivly) huge delta in interrupt routine delay. (like needing one more register on stack, or no stack at all, or having a constant as a LDR or as a MOV)

It turns out after measuring that the worst case C64 interrupt delays all happen during a read of the SDIO->STA register. Reading this register takes 7 or 8 cycles (not documented anywhere but I measured this with the debug cycle counter).
But the jitter I measured in the interrupt can be more than just those 7 cycles, if the ARM's write-buffer is still not empty. So the worst-case latency that came up in my measurement, was the SDIO register write (takes around 6 cycles) followed by branch to a load of the SDIO->STA register (7 cycles). Total latency around 15 extra cycles on top of the minimum 12. 

It seems due to the non-empty write buffer, but the LDR already being in the pipeline of the ARM, the whole chain cannot be aborted to handle the interrupt any earlier.

@Kim would you be interested in a pull request for the sdio fixes I made?

KimJorgensen, 2021-02-25:
Thank you for your detailed description.

No offense, but to avoid confusion and respect the license, I would appreciate if you distinguishes between Kung Fu Flash and your derived work. As far as I know, Kung Fu Flash does not have any stability problems when accessing the SD card.

It was a conscious decision not to enable SDIO HW flow control due to the problem described in the [errata sheet](https://www.st.com/resource/en/errata_sheet/dm00037591-stm32f405407xx-and-stm32f415417xx-device-limitations-stmicroelectronics.pdf), see section 2.13.1.

I would be a bit reluctant to accept a pull request that fixes a non-problem, especially with the above in mind, but thank you for the offer.

medzes, 2021-02-25:
Ah thanks, I had not seen that errata sheet. In that case the hardware flow control should be left disabled. The 1/4 fifo trick still helps sufficiently for my code.

Agreed, if the SDIO issues are a unique result of the set of pin assignments I am using and possibly different gcc version (I use the one from ubuntu 20) that is fairly strange but no need to merge.

No offence, but I am not sure how to work along with your request on license. I don't think I have sufficient understanding of legal issues to understand the request you have.

I would not expect the changes I made that I have published so far to be allowed to be called a different project. The only substantial change is in one file (c64_interface.c) to allow two optional choices of GPIO pin assignments during compilation and other minor changes to allow -Os compilation. I've personally only seen this as minor software changes to allow support for a "hardware prototype" version of KungFuFlash.

It appears you have far more experience with that than I have, would you care to offer a suggestion how to proceed?

The other changes so far are adding soft-kernal support, optional LCD, and optional side-load of ARM code. Those features use some additional files which I have used the same copyright as yours, but with my name as author. I suppose including those in a compiled binary could be called a derived work, indeed. Would you be happy if I called that KungFuFlash-devebox, to clarifify it works on the devebox prototype ?

KimJorgensen, 2021-02-26:
As long as it is clear that your work is not the original Kung Fu Flash I'm happy.
I'll close this issue and thank you for the info.

closed 2021-02-26 by KimJorgensen